jotego/jtcores #3: how insert coin??

vastago, 2019-01-26:
After create rom and put with the core in mist, is great to see the game presentation and insert coin message but, how in the world is possible to start the game??? Dont work with joy in usb mist, megadrive pad, o any keyboard keys... I dont understand.  it is necessary to have a keyboard with numeric keypad maybe???

jotego, 2019-01-26:
Everything is mapped to the gamepad. I must check if this is properly explained in the README file. The first two gamepad buttos are jump and fire and the other two are coin and 1P start. I have been [asked](https://www.patreon.com/user/community) to add keyboard support compatible with MAME and I plan to do it.

closed 2019-01-26 by jotego

vastago, 2019-01-26:
Thanks for the clarification, it has been very useful for me. Tomorrow I'm going remapping the usb pad because I dont think the genesis / megadrive pad work. Good work mate.