jotego/jtcores #6: The first (and I think the second) is always the torch.

manast, 2019-02-12:
This is something that bothers me from MAME as well. Maybe I am wrong and I do not recall the original arcade, but I am quite confident that the items were random. However in this version we get always a torch as the first item, and I think as second item as well, third item the knife. Maybe there is some issue in the way the random number generation is performed? maybe it is using a static seed?
In any case, it hurts the game a bit, since if you take the torch by mistake in the first stage, you are more or less smoked :D.

manast, 2019-02-12:
ok. So I have been watching some youtube videos from the real arcade and surprisingly enough the torch is always the first item. This really confuses me, maybe the arcade I used to play in Spain was a bootleg?

manast, 2019-02-12:
I made a discovery that solve the mystery. The torch is indeed the first item, and the second item, but only the FIRST time the machine is played. So probably the random number generator is based on a seed that is always the same at start, but in subsequent games you may or may not get the torch as the first object. As a kid obviously I did saw and played the game long after the machine was turned on, and therefore I do recall that the torch was not always the first item :).

jotego, 2019-02-13:
This is just the way the game is. The random generator seems to advance only during game play and not during the demo. You just cannot make pure random numbers with purely digital signals. Modern processors have an analog noisy circuit to generate random numbers.

closed 2019-02-13 by jotego

manast, 2019-02-13:
yes, I know how pseudogenerator number generation works. Even in a digital circuit it can be generated a seed for example based on the amount of microsenconds that has passed since the machine was started. Clearly this is not the case with GnG, they probably just use a manual counter, which is quite interesting, since you should in fact get a deterministic play if you the player played exactly the same from a clean boot of the machine.