jotego/jtcores #7: Strange noise in the second stage

manast, 2019-02-12:
in the second stage (ice castle) there is some strange background noise. Its almost like a one of the platforms makes a noise that fills too much, not sure.

jotego, 2019-02-13:
If you play the game on MiSTer through HDMI you may be getting a resampling effect due to the difference in audio sampling rates of the original hardware compared to HDMI. Try using directly the analog audio output.

closed 2019-02-13 by jotego

manast, 2019-02-13:
This is on a MiST, using standard analog output. I am using the latest image from 9th Februari, latest MiST firmware as well. The noise is quite loud, should be easy to reproduce it happens every time on the ice stage.

jotego, 2019-02-14:
Could you provide a video of this?

manast, 2019-04-02:
Ok, it took me a while but here is the video anyway: https://s3.eu-central-1.amazonaws.com/castmill-medias/IMG_6538.MOV
What do you think? I do not remember this strange sound on the original arcade, but it was long time ago...