jotego/jtcores #8: Smooth scrolling

manast, 2019-02-12:
This is something I do not recall from the original arcade. I know that in the first versions of MAME (1998 or something), the scroll was super smooth. However, at some point the scroll stopped to be smooth, now, it could be that the correct emulation is the latest version, i.e. with non-smooth scroll. Since this is a hardware recreation I guess that it is correct like this, but since you own the original hardware, could you just confirm this fact so that I can go to sleep with peace of mind? :).

jotego, 2019-02-13:
Unless you use a real tube TV you will never get a perfectly smooth scroll because your screen has to convert between the arcade resolution and your screen resolution and that creates that feeling with the scroll.
When you played the first versions of MAME, you may still had a tube monitor rather than a flat one.

manast, 2019-02-13:
Yes I have played both in CRT and LCD monitors. I do not think this is the issue here, however It may be that the original game has not a smooth scroll, this is easily comparable with Black Tiger, which I think uses a very similar hardware, still the background scroll is super smooth (in CRT or LCD).

jotego, 2019-02-14:
The 15kHz video generation of this core is exactly the same as the original hardware, part by part.
If you use the VGA or HDMI outputs or if you go through a screen that requires a resolution change, then that will affect the feeling of smoothness. 
However, the 15kHz output on a CRT screen that does not add image processing, should look exactly as the original. 

I do not have such a screen to verify it but the core has been designed matching the electrical schematics of the PCB so it should be the same. I will double check in case there is something wrong in the translation from schematics to Verilog.

Santabomber, 2019-02-14:
I will see if I get some time tonight or this weekend to test. I have several GnG clone pcbs (sadly sold my original long time ago, but clone is spot on!). I can only try on CRT / Arcade cab, I can not connect to any lcd...

jotego, 2019-02-14:
@Santabomber please let us know the results of the original hardware on CRT. The term "smooth scroll" is a bit vague but maybe you can compare with the MiST/er version too.

jotego, 2019-02-16:
Just a note: scroll is smooth during the map screen.

jotego, 2019-08-18:
Scroll hardware is smooth as verified in other cores -like Commando- which use the same video module. It is the game software which moves the scroll in that way.
Closing the issue.

closed 2019-08-18 by jotego

jotego, 2019-08-25:
I have been using the MiSTer scan doubler and noticed that it has a distorting effect on the scroll. Maybe that is the problem that was originally reported. This was not seen by me on MiST, which is my main developing platform. 
I will try to use mist scan doubler on mister too for next release.