jotego/jtcores #15: Wrong jtframe?

DFL1973, 2019-06-26:
Hi jotego, thanks for sharing your work.
I'm trying to build the GNG for Mist, but I' getting a few errors at the jtframe part. It looks like the GnG is using some older version, with fewer parameters. Can you point me the correct one, please?

jotego, 2019-06-26:
Hi,

Sorry it is currently broken only working for 1943 core. I have to spend a
couple of days updating it.

If you move back to an older version of jt_gng repo and update the jtframe
submodule accordingly it will work.

I may not be able to fix the current repo at least for the next 2-3 weeks.
I should have worked on a branch, my fault.

jotego, 2019-08-31:
Hi @DFL1973 

This is fixed now. I hope not to break it again.
Also if you can compile GnG now you will get a better version than the published one as it has already some improvements made to it.

Thank you.

closed 2019-08-31 by jotego