jotego/jtcores #24: 1942 graphics

DFL1973, 2019-10-07:
Hi Jotego. 

I just built 1942 from the last source you published for the MiST and there are a few graphical bugs. The background is shifted about 8 pixel and there are some missing parts on the sprites (or shifted sprites).

The other games are fine.

Thanks for sharing!

jotego, 2019-10-07:
Hi,

I’m about to update the repo with all this fixed. Stay tuned please. :-)

Thank you

jotego, 2019-10-08:
Fixed with [current](https://github.com/jotego/jt_gng/tree/0542b1241d7b00161737ee7f5cfd4c6f67c31247) release.

closed 2019-10-08 by jotego

DFL1973, 2019-10-08:
Confirmed, thanks!

A semi-related question: Is the Vulgus using the same 1942 synth, just changing the ROM? Or maybe I just didn't understand how to build Vulgus.

jotego, 2019-10-08:
Just follow the instructions using jtcore command. That will generate the
qpf/qsf files. Then you can inspect them.

Vulgus and 1942 have very similar hardware but you cannot just swap the
ROMs. They share the same codebase but a macro definition selects different
pieces of the code for each core.