jotego/jtcores #25: MiST phase shift set to 180º

gyurco, 2019-10-07:
I've noticed this commit, and want to make a note: if you're using the correct PLL for the SDRAM clock, you won't need any phase shift until about ~100MHz. Just add this to the .qsf:
`set_location_assignment PLL_1 -to "pll:pll|altpll:altpll_component"`
of course set the correct pll name, the main point is to use PLL_1, which has a dedicated output to the SDRAM_CLK pin.

jotego, 2019-10-08:
I tried it but it didn't work. I think that when if I align the two edges, data will not come at the expected clock cycle in the SDRAM controller. There is a wide range of shift values that work but zero has never worked.
I always find this puzzling because as you know the tool will delay clocks all over the place and if you also add a shift to this other clock the question about which edges to align arises. I think that with zero shift, the tool (compiler) tries to align both edges (internal and external) whereas with other values, it doesn't.

gyurco, 2019-10-08:
As I see, the sdram controller doesn't use the on-pad output registers (fast output registers). You can see this in the fitter report of the output/bidir pins (output registers: off). Together with fast output registers and PLL1 used for SDRAM clock, there shouldn't be a need for phase shift < 100 MHz (all current cores works this way, like MiSTery, with 96 MHz SDRAM clock). And another story that the C64 core used 180 degree phase shift @64 MHz, but when I switched to PLL1, it became unstable, and required to change back to 0 degree.
Also it's possible that not all PLL outputs are suitable for the dedicated output, c0 and c3 are surely OK to use.

jotego, 2019-10-08:
How do i enable the on-pad registers?

gyurco, 2019-10-08:
Basically this is needed to use a 0 degree phase shift (to use 96 MHz, another NOP is needed between ACTIVE and READ/WRITE)

```
@@ -178,7 +178,7 @@ always @(posedge clk)
         if( cnt_state!=3'd0 ||
             (!downloading && (read_req || refresh_ok) ) || /* when not downloading */
             writeon   /* when downloading */) begin
-            if( cnt_state==3'd4 || (autorefresh_cycle&&cnt_state==3'd3) )
+            if( cnt_state==3'd5 || (autorefresh_cycle&&cnt_state==3'd3) )
                 cnt_state <= 3'd0; // Autorefresh needs only 60ns
             else
                 cnt_state <= cnt_state + 3'd1;
@@ -247,13 +247,13 @@ always @(posedge clk)
                 autorefresh_cycle ? CMD_NOP : CMD_READ;
             data_rdy    <= 1'b0;
         end
-        3'd3: begin
+        3'd4: begin
             if( read_cycle) begin
                 data_read[31:16] <= SDRAM_DQ;
             end
             SDRAM_CMD <= CMD_NOP;
         end
-        3'd4: begin
+        3'd5: begin
             if( read_cycle) begin
                 data_read[15: 0] <= data_read[31:16];
                 data_read[31:16] <= SDRAM_DQ;

gyurco, 2019-10-09:
I've played a bit with the SDRAM controller, and instead of a negative clock, I've changed it to a 96MHz one (effectively it'll be the same). This should be super-stable, at least on MiST it is so far.
https://github.com/jotego/jt_gng/pulls
https://github.com/jotego/jt_gng

If you're using the mt48lc16m16a2.v with Verilator: the # notation is not supported by Verilator, and maybe you'll get the SDRAM simulation data arrive sooner than in reality.

jotego, 2020-05-16:
I finally made this in a different way but did use your code as a reference. Thank you.

closed 2020-05-16 by jotego