jotego/jtcores #33: Blacktiger and Tiger Road horizontal scroll not completely smooth

manast, 2019-12-30:
Hello,

I just noticed that the horizontal scroll in both Blacktiger and Tiger Road is not completely smooth, there is something that looks strange, can be compared to the vertical scroll which is perfectly smooth in both games, however you will need a CRT monitor to notice this, on a LCD due to slower screen update and ghosting you will not see any difference.

manast, 2019-12-30:
Ok, I also noticed that the map scroll in GNG that was completely smooth before now exhibits the same non-smoothness as the above mentioned games, so maybe it is a regression on some generic code that affects all games?

jotego, 2019-12-30:
Hi,

If you play on a CRT, unless MiSTer frameworks is doing something with the video, it should be as the original and the original isn't always smooth. But I agree that GnG map should look smooth. I haven't modified anything related to that in a long time. And it would be difficult to create that effect, to be honest. The scroll is rendered in real time, no buffers. Not even a line buffer. 
However, if you play on a modern screen, it is possible to get scroll jumps because of the scan doubler. There is a line buffer in the scan doubler and that can create this effect. In MiSTer, a filter after the scan doubler usually removes that effectively.

manast, 2019-12-30:
Thing is, the GNG version from 9th februari has smooth scroll, whereas the latest version does not have it, and I am using the same MiST for both tests (so it cannot be a firmware thing). I will try to make a video but I do not think you will be able to see it, but trust me on this, its smooth and the latest version is not :)

manast, 2019-12-30:
So to make it clear, this is on a MiST, not MiSTer, not sure that makes any difference at all anyway.
And the effect is not as if the scroll is bumpy, it looks as if it was an LCD screen instead of CRT, like it has some kind of blending of the previous pixel with the current one while doing the scroll, you can see it without any doubt on a CRT screen if you have one to test it. The strange thing is that on Blacktiger and Tiger Road, the Vertical scroll is smooth, it is only the horizontal that suffers of this issue.

manast, 2019-12-30:
ok, so I tried to capture this on camera, even with slowmo and there is no way to show the effect, it is blurry in all cases... but as mentioned, just check on a CRT and you will see it 100% sure.

manast, 2019-12-30:
Maybe this can be a hint on what is going on. Basically what I have noticed is a very subtle difference between the old core and the new one. In the new one, seems like the pixels are not of the exact same size in the horizontal axis, and this seems to affect the scroll too somehow. Note that this same problem appears in Blacktiger and tiger road, maybe other cores too I haven't checked them yet.

This is the old core (February):
[image]

And this is the new one, notice how the zeros are not identical as they should be:
[image]

jotego, 2019-12-31:
Can you tell me specific build dates (the date in the rbf file name) for two versions of the same core, one with the problem and one without?
I assume you are using then:
*MiST
*A tube 15kHz screen

manast, 2019-12-31:
Version without the problem: gng_mist_20190209
Version with the problem: jtgng_20191215 (from jtbin).
I may have some time later today to do a bisec and find the exact version that introduces the issue.

manast, 2019-12-31:
Ok, so I could verify that the problem is also from version jtgng_20190914
however, when I test the other older versions 190221, and older, I get crippled graphics, maybe the rom is not compatible with the older cores?

jotego, 2019-12-31:
Are you using a tube VGA screen or a 15kHz RGB screen? 
(BTW there is a new Black Tiger RBF for MiST in Google Drive)

manast, 2020-01-01:
15KHz RGB screen

manast, 2020-01-02:
I tried the new core. The glitches are gone :), however there are a couple of new ones. When the sprites are too close to the left border it looks like this:
[image]

On level 5 there is also a strange graphic glitch on the top:
[image]

Finally, regarding the scroll. I think the problem is related to the odd scroll positions. In blacktiger when you move the character you cannot just move it one pixel, I don't know how many it moves, maybe 4 or 8 at a minimum, but this is an even number, so when you are not scrolling the background looks good. However, during the scroll, the odd positions are wrong, not sure if it is that they are 0.5 pixels off, or that not all the pixels in one horizontal line are of the same size. This makes the scroll look harsh. Would this ring any bell on what it could cause this effect?

jotego, 2020-01-02:
In these games, the CPU writes to a registers containing the start position for the background memory scan. The CPU typically only writes to it during vertical blank. The memory is read and the data is output to the analogue video without any filtering at all. Notice that there are 256 pixels in the image whereas your CRT screen will have a larger number of "pixels" or display units so you will experience some "resampling" effect. This is more noticeable if the quality of the signal is good (RGB components with good bandwidth). Have a look at this [video](https://www.youtube.com/watch?v=x0weL5XDpPs) to see how analogue signal artifacts affect the image.

I think that you before you were just lucky with how the 256 pixels fell on your screen. At some point there might be a shift of one or so pixels that make them fall in a less good position. Try adjusting the horizontal position setting on your CRT.

manast, 2020-01-02:
I am quite sure this is an issue with the core. I have quite a lot of experience coding smooth scrollers, and this is definitely an issue. Furthermore, the old GNG core has perfect scrolling, and the same for megadrive, snes, amiga, cores, etc. 
I think the issue introduced after the last known working release the 9th of February introduced some artifact that affects both the horizontal scroll as well as the rendering of some graphical elements such as the score, etc.
If you could use an 15Khz output and compare the scroll from blacktiger and the original board you will see this problem right away...

manast, 2020-01-03:
I did a final check. The scroll issue, etc in GNG was introduced in version 20190914. Last version I can verify it works correctly is 20190221. Interestingly, that version also seems to be the version that requires a different ROM format, the ROM I have for 20190221 does not work on 20190914 and vice versa.
It would be great if you could check the diff between these two versions, it cannot be a very big thing what causes this...

jotego, 2020-01-12:
These two games are reported as smooth on MiSTer setups with CRT.

manast, 2020-01-12:
Or maybe they do not notice the effect :). Unfortunately I do not have a MiSTER to test, but I tested with standard LCD today and the effect is in all games (after 20190221 GNG AFAICT).  So I am sure you can also reproduce it in LCD it is just that it is far less visible, and can be confused by the non smooth nature of scrolls in LCD due to ghosting, and also because the games are not in the native resolution of the LCD screen. BTW, I see this is not only for background but all graphics rendered, depending on the coordinate the pixel gets more or less thick.
In Double Dragon on CRT it can be easily reproduced in the first level, just in the beginning, there is a long vertical line, depending on the scroll position this vertical line gets thicker or smaller. The problem is that DD does not work on my LCD due to the non-standard refresh rate, but I can post an image of this effect for the CRT at least, see if MiSTER users can or cannot reproduce it.
I know I am being picky about this but it must be perfect right? :). And MAMe for instance gets this particular thing right.

jotego, 2020-01-12:
Check the new DD files for MiST the ones dated for today. You should get video now correctly. Take a shot of that vertical line and we will compare with MiSTer.

manast, 2020-01-13:
Here comes the screenshots. In this case it is quite clear that one of the vertical lines gets thicker will the other gets smaller. This same effect happens everywhere, so when scrolling it creates a weird effect instead of smooth evenly scroll.

[image]

[image]

jotego, 2020-01-13:
This could happen if the pixel signal is being resampled at some point. There should be no resampling for 15kHz video output. I had a first look yesterday through the signal chain and I didn’t find it. But I’ll look again more thoroughly.

manast, 2020-01-13:
Remember that the issue was introduced between GNG 20190221 and 20190914, so it maybe helps to can check for suspicious commits during that time.

manast, 2020-01-13:
also that it must be some generic piece of code since it affects all cores (or at least all horizontal cores, I have not tested vertical cores).

paulb-nl, 2020-03-07:
https://github.com/jotego/jtframe/blob/affbd44fd4d34090e8a825e31dacdd57893415dc/hdl/mist/osd.sv#L225-L227

There it uses ce_pix as clock enable for R/G/B_out but ce_pix is generated by the OSD and is not synced to the game's pixel clock enable. In this case it will output a pixel every 6 clocks instead of every 8.

jotego, 2020-03-07:
Thank you @paulb-nl ! That's the resampling point that was causing the problem. I have uploaded the fixed cores to [jtbin](https://github.com/jotego/jtbin/commit/e1f3cd38709947ccf855b94e4c86e7b708680bd4). The change in the source code is already [published](https://github.com/jotego/jtframe/commit/43a325df2996b30aabae02920b0abcb5d1bea449) too.
Once @manast confirms that it works for him I'll close the issue.
Thanks again!

manast, 2020-03-07:
Yes yes and yes!. It works. I tested with black tiger but I am sure it works with all the other cores too. I am Super happy 🗡

jotego, 2020-03-08:
Great news. Sorry I didn't notice before and thank you for insisting on this problem. @paulb-nl thanks again!

closed 2020-03-08 by jotego