jotego/jtcores #38: Double dragon sync issue with CRT on MiST

manast, 2020-01-11:
Hello,

here a video that demonstrates the effect on my CRT. My CRT works well with NTSC signals (60Hz), and PAL (50Hz), so I think there should not be any problem with 57Hz either.
Would be interesting to know if anybody has managed to get it work correctly on a CRT using the MIST (the MiSTER seems to work well).

https://drive.google.com/file/d/1b1BOXyalkyoc80vPWyyEpzXs-97Rf-Ff/view?usp=sharing

jotego, 2020-01-11:
Please try the file jtdd2_20200111b.rbf in Google Drive. I have widen out the sync pulses. Let's see if that helps.

manast, 2020-01-11:
Ok, I was referring to DD1, and since there is no .sh to convert the rom I could not test DD2, however I tried to load the core and the OSD menu is being displayed correctly, so I think the widening of the sync pulses did it, could you release DD1 with the same fix?

jotego, 2020-01-11:
Yes, that's a very good sign. I have uploaded the buildrom.sh which generates the ROM file for both games. I am compiling the DD1 version with wide pulses now and it will in Google Drive in 15 minutes.

manast, 2020-01-11:
Amazing, thanks!

jotego, 2020-01-11:
The new DD1 core is called jtdd_20200111.rbf

manast, 2020-01-13:
Works almost perfect, one small "wave" kind of glitch on the top though:
[image]

jotego, 2020-03-02:
Hi Manuel,

I have got one more report similar to yours about CPS. I will be checking once more the MiST video signal path to try and figure out what's causing the issue.

Stay tuned ;-)

manast, 2020-03-02:
yeah, I was actually counting the days for you to finalize CPS so that you could find time to look into this :)

jotego, 2020-03-08:
I close it now. Double Dragon public version -to be publish this month, I hope- will incorporate the fix we have already tried on CPS0 cores.

closed 2020-03-08 by jotego