jotego/jtcores #40: Last build with black screen

vtrucco, 2020-01-27:
Hi Jotego,

I just built from the last sources (stable branch) for MiST and I'm getting black screens on Bionic Commando and Tiger Road. Black Tiger is working and F1 Dream fails to build.

All games were built following the instruction on README.

The pre-built cores are working
https://github.com/jotego/jtbin/tree/master/mist 

Thanks for sharing!

-----------------------------------------

Critical Warning (10191): Verilog HDL Compiler Directive warning at jtframe_mist_top.sv(75): text macro "CORENAME" is undefined
Error (10170): Verilog HDL syntax error at jtframe_mist_top.sv(75) near text ",";  expecting an operand
Critical Warning (10191): Verilog HDL Compiler Directive warning at jtframe_mist_top.sv(325): text macro "GAMETOP" is undefined
Error (10112): Ignored design unit "jtf1dream_mist" at jtframe_mist_top.sv(27) due to previous errors

jotego, 2020-01-28:
Hi,

Thanks for trying this out.

Sometimes going into a module, particularly jtframe, and getting the latest commit is needed. Git can keep track of the required commit needed for a submodule, but sometimes I am not very good at doing that, particularly as I am always editing something in these repos. Try going to jtframe folder and typing: `git checkout unstable` to get the latest commit.
About F1-Dream... I suppose you are using linux to build this, right? If you aren't F1-Dream will not work as it relies on symbolic links. Be sure to be using the most recent commit of the **stable** branch of jt_gng.
If you get a blank screen just press F12 to load the ROM. I don't know why sometimes MiST fails to autoboot the games.

closed 2020-01-28 by jotego

vtrucco, 2020-01-28:
Hey, thanks for the reply.
Tried with the "unstable" jtframe, but still with black screen. Same as loading the rom with F12.

I will try to update the other modules to see if if it helps.

Thanks!