jotego/jtcores #41: Black Tiger hangs up after completing the second level

jotego, 2020-02-13:
-Seen in MiSTer
-Not sure of which ROM version

Hewhoisred, 2020-02-13:
I'm guessing this can happen at the end of any boss fight.  For me I've seen it hang at the end of level 4, 5 and 6.  When it hangs, it happens right as the protagonist goes into the kneel pose before he jumps up out of the top of the screen to go to the next level.

When he looks like this:
https://youtu.be/IA7fFTttvyg?t=1168

I will try and figure out what rom version I'm using.

Chocobobobo, 2020-02-13:
I had this issue happening to me three times and was about to open a new issue.
The game would freeze just after beating a boss (I had this issue only with dragons).
So it is happening not only in Stage 2. The game would freeze in this position:
[image]

 I don't have 100% repro steps, but now my heart is beating faster after defeating a boss... Will it freeze and force me to reset or not? Such suspense... I will add more details if I can have some proper repro steps. I am playing Black Dragon.

Hewhoisred, 2020-02-13:
Yeah, whether or not it hangs appears to be completely random, but at least it always happens right after you beat a boss.

jotego, 2020-02-26:
I think this has to do with noise in the SDRAM tracks. An issue present in all release cores but that is sensitive to the data on the bus. So not all games have suffered it. It became very obvious on the CPS beta release and I found a solution once I understood it.
I have [updated](https://github.com/jotego/jtbin/blob/master/mister/btiger/releases/jtbtiger_20200226.rbf) Black Tiger today with that robustness improvement. Please let me know if that fixes the random hangs.

You can get the update through the update script or download it from the link. I close the issue for now. Feel free to reopen it if needed.

Chocobobobo, 2020-03-03:
After playing much more of Black Dragon, I am sorry to report that the game is freezing in new random spots now. Here is an example where the game froze (beginning of second stage)
Music will keep playing; but except that, everything is frozen.
[image]

So right now with the latest core we have:
-Random freezes (it is a new issue compared to older version)
-Freeze after the game over screen at the STOP - WARNING screen (this issue has been around for a while)
-No freeze after beating a boss so far (Potential fixed issue compared to older version)

Hewhoisred, 2020-03-03:
Yep, it's still crashed after a boss kill for me, happened at level 7 boss just the other day.
I have not experienced any other crashes, yet.

Chocobobobo, 2020-03-04:
Ouch... Would be good to reopen this issue then...

jotego, 2020-03-04:
Could you please try the bootleg version? There is an important difference with the original: the absence of the microcontroller. If the bootleg doesn't crash then it points me towards debugging the MCU.
Thank you!

Chocobobobo, 2020-03-07:
I have been hammering the bootleg version as much as I could.
[image]
[image]
[image]

And you might be on something when you mentioned the MCU... I had no crashes at all, and it feels more stable overall. Of course, I will keep playing it and let you know if I encounter any crash.
Good luck with the MCU debugging (current only potential criminal apparently) and don't hesitate to throw any new / test version for us to check!

jotego, 2020-03-08:
Thanks a lot. I'm going to investigate the MCU problem. This is a very important hint!

Hewhoisred, 2020-03-08:
I just now played through the bootleg one time, did not crash.  I will do a few more complete runs and post to let you know.

jotego, 2020-03-08:
I have made a new commit that fixes the MCU timing. Please update and play with the original version now. The version is jtbtiger_20200308.rbf

Chocobobobo, 2020-03-08:
Thanks for the update. Sadly, I have to report the game (original version) froze again with the 0308 version.
Once after I beat the last dragon; the game froze, but the music kept playing.
[image]
I will test it more, but wanted to report the issue asap.

jotego, 2020-03-09:
Thank you. This looks like a bug inside the MCU implementation. I don't think the author of that IP is available and it is very hard to debug it. I'll leave this issue open until I find a solution (a different MCU IP or I build a test suite to debug it).
Meanwhile we know that the bootleg (no MCU) version works and that the original has random hangs.

Chocobobobo, 2020-03-09:
Right; that is correct.
Let me know if you want any extra testing to be performed on anything you see fit.

jotego, 2020-03-19:
After much debugging of **Double Dragon 2**, eventually I found that a different Z80 IP does not have the issue that was holding me back. I have seen that behaviour from the regular VHDL T80 IP that I normally use before so I think it could be related to this issue too.
I have compiled a Black Tiger version using the other Z80 IP. You can get it with the jtupdate script (MiSTer only). Please let me know if this one works fine with the original ROM.

Chocobobobo, 2020-03-19:
Hello,  I was about to send an update and I saw your comment.
Sadly, the original rom is still freezing with the 0319 build.
Same scenario as usual; I beat a boss; game freezes; but music keeps playing.
[image]
Seems like we are back to the original MCU suspect / most likely criminal.

Chocobobobo, 2020-03-25:
Quick question, will this core also get the 96MHz memory controller update?

jotego, 2020-03-25:
It is not needed for this core. I will leave it as it is, running with the controller at 48MHz.

jotego, 2020-09-11:
@Chocobobobo There is a slight change that the issue could be cause by some idle cycle in the CPUs. As I said, slight change. If will make a new RBF if you are willing to try ending the game again. Just asking!

Chocobobobo, 2020-09-20:
Sure thing.

Chocobobobo, 2020-10-11:
Let me know when you have two seconds to upload the update .rbf you want me to check somewhere.
And I know it is not like you are bored and have nothing to do :)

Chocobobobo, 2021-02-12:
And the issue is still open in today's update of the core. Just wanted to let you know :)

[image]

jotego, 2021-09-04:
Fixed in 37de6192

closed 2021-09-04 by jotego