jotego/jtcores #44: HDMI Aspect ratio

paulb-nl, 2020-03-04:
The current aspect ratio on HDMI is 4:3 but the video on a CRT is quite narrow and closer to 5:4 because of the 384 pixels per line with only 256 active pixels.

jotego, 2020-03-05:
I think I have to disagree on this one. Aspect ratio on CRT games does not depend on the number of pixels per line. Have a look [here](https://www.youtube.com/watch?v=LHfPA4n0TRo).

paulb-nl, 2020-03-05:
That is incorrect.

Compared to SNES/ Genesis or CPS1 the image is much narrower on my CRT with CPS0 because of the active pixels are only 66% of the total line count. Normally this is 75%. Gng has black bars on the sides for example.

The only way to show it 4:3 was if the CRT was calibrated to fill the screen for CPS0 games. Of course this can be done on arcade cabinet with only one game. I am curious if a real Gng machine is calibrated or has black bars on the sides.

jotego, 2020-03-09:
The blanking period of these cores should be accurate but I am not so sure about the sync signals. They are not used by the game so I may not be using the original timing for VS/HS. I should check it. But as I said, VB/HB should be accurate.
I see your point. Once there is a knob someone can move there is no absolute truth for the aspect ratio. Note that in that video there is a horizontal bar too at the bottom...
I think this can probably be made configurable in the OSD for MiSTer. I'll look into it (or you're welcome to try modifying jtframe to support it).

M-Walrus, 2020-04-07:
I would also like to see this fixed. The aspect ratio is most certainly incorrect over hdmi. I have a crt right next to my PC and it is quite apparent. Characters are wider than they should be, etc.

Another big thing that this affects is the excellent filters we have on MiSTer. All of the interpolation filters end up blurrier than normal, because the image is wider than it should be.  This is especially damaging to the image when using the scanline filters, since they have interpolation baked-in and they will all end up looking much too blurry, compared to the expected result that you would see with other consoles and arcades on MiSTer.

jotego, 2020-04-14:
I think this should affect most CPS0 cores, as they share the same video timing (pixel clock, blanking signals, etc.)
I have updated all cores except Bionic Commando, which is pending MCU maintenance. It's hard to judge whether this is 100% correct or not. I guess if we could identify circles in the games they would become the reference we need.
Anyway, I'm closing this. Just run the update script to get them.

closed 2020-04-14 by jotego