jotego/jtcores #45: Garbled sprites

the1ross, 2020-03-07:
The older Jotegi cores work fine but new ones like Tiger Road, F1 and Bionic Commando don't work properly for me. Many of the sprites are either garbled or don't appear at all. 

I've not had this problem with any other cores including all the arcade ones. I was hoping that after an update or two, things would fix themselves, but they haven't. Any advice would be appreciated. 
[image]

the1ross, 2020-03-07:
In this one enemy sprites among others don't appear and bionic command disappears. 

[image]

the1ross, 2020-03-07:
F1

[image]

the1ross, 2020-03-07:
Black Tiger is the only newer core of you're that works perfectly. Gng, 1942, 1943, gunsmoke etc all work fine also.

jotego, 2020-03-07:
I close this thread. Let's track this problem [here](https://github.com/jotego/jtbin/issues/4).

closed 2020-03-07 by jotego