jotego/jtcores #52: Bionic Commando MCU/inputs issue

jotego, 2020-05-20:
Bionic Commando fails to read the inputs. The problem started when I moved the 8051 code to use clock enable signals. The 8051 was problematic with sound and one reason might have been not meeting timing.

Once the 8051 used a clock enable the whole design was cleaner. F-1 Dream uses the same MCU and worked fine. However, Bionic Commando input stopped working. For Bionic Commando the MCU has access to part of the input bits but not to all. Communication about input is likely to be done through main memory.

The same 8051 is used in Black Tiger, and there is an issue about it (https://github.com/jotego/jt_gng/issues/41). Bootleg versions of Black Tiger work ok.

For Bionic Commando I have been able to play with the bootleg version at the time I was preparing commit https://github.com/jotego/jt_gng/commit/c1076b9aed1bc7f0eeed28831eb7e53351bbdc84 but I am not sure whether the commit works for the bootleg or not. The bootleg is not supposed to have an MCU.

There is no nice 8051 IP available for retro purposes. All IP available comes from people who want to make stuff roughly compatible with 8051 but all I have seen -even those coming from private companies- do not respect the original chip interface or peripherals. Let alone have a clock enable input.

So there is quite a bit of work to do to debug what is wrong with either the 8051 IP or the interface with it.

The core status, other than that, is up to date with JTFRAME and simulation works.

gyurco, 2020-09-04:
One thing noticed btw. The core doesn't accept the coin input, however it's not read by the MCU at all (coin input is cabinet_input[15:14], the mcu gets only [7:0]).

jotego, 2020-09-04:
Thank you very much for your insights. 

When I changed the MCU clock speed the core stopped working correctly. But I'm not so sure the problem was that. 
Sound used to be broken too but with SF1, which uses a similar sound subsystem, I had a similar issue and I think it came down to initial values of the clock enable counters. Once I fixed those values it seems to always compile correctly. 

The MCU is also a suspicious part but F1-Dream uses the same one and works.

I think there is something wrong in the way the M68000 is hooked up. I should compare it with other cores using that CPU that are working correctly.

jotego, 2020-09-06:
I tried your https://github.com/jotego/jt_gng/pull/58 locally. It works. But then if I merge that into the SF branch -remember you have access to jt_gng2 too- it doesn't work.

All jtbiocom_* are equal but there are some changes in video related files in the common modules/jtgng_* path due to minor changes made for SF.

That to me points to a problem with the compiler as the files that change are not in the sound subsystem, not even in the top level. The files affected are all in the video subsystem.

Anyway, thanks to your fix, the one in stable works -until I merge the branches- so I can release an update of the RBF with the new framework. That's good news! Thank you :-)

jotego, 2020-09-06:
I have now updated all RBF files in jtbin. Thanks again for making this update possible.
I may try using yosis to generate a different simulation set of files, or maybe even a different set of input files to Quartus and see if that brings some light into this.
I hear yosis can take VHDL produce C++ or verilog outputs... I have never used it though.

gyurco, 2020-09-06:
Great! Bionic Commando was held for a long time, good it's finally works. I encountered minor video glitches in some places, is it possible that video data from ROM doesn't arrive in time in some cases? At least it reminds me similar issues in other arcade cores which I hacked on.
Access to jt_gng2 doesn't work for me, if this is the correct path:
https://github.com/jotego/jt_gng2/

jotego, 2020-09-06:
I thought I had sent you the invite last week... I've just sent it again. Please do not distribute files from it (or RBF files). I eventually advance the public stable branch from this repo each time a core gets good enough.

jotego, 2020-09-06:
Yes, there are some minor graphic details to fix. The extra vertical line on the RHS and originally there was a bit of sprite flickering at the beginning of one of the late levels. The flickering may be gone by now as the object engine has seen a lot of changes since the Bionic Commando days. The vertical line issue might be a matter of adjusting the HOFFSET parameters or the blanking. Worst case would be that the scroll pixel delay is too large. That would require more work.

gyurco, 2020-09-09:
Hmm, just compiled Bionic Commando from jt_gng2, and it just works as the one in jt_gng.

jotego, 2020-09-10:
I just tried again and it failed. 
I used commit: 4522df34f3383f
And git submodule:
 026efe45ceab6fedc551ee05d69b2d0a60b0ba53 fx68k (heads/master)
-56ee1717e261271bb010be52daadeed345bf9ac9 jt12
 75a705762439585efd4a059813baaf58081c7349 jt51 (v1.2-43-g75a7057)
 4712ab65ab5eb0526d7440d00d60b4a1a604efa8 jt5205 (heads/master)
 cb5946a23399a5f1a0bde45d826e209492f9d626 jtframe (toraok4-193-gcb5946a)

I don't get sound. Can you confirm that you get sound with these commits?

gyurco, 2020-09-10:
It's silent. The last one worked is: 6964af37
I've copied jtbiocom_sound.v from jt_gng, and also works (everything else is 4522df34f3383f).
As I see, you changed the cs signals to registered ones, maybe this introduces a delay which breaks something.

jotego, 2020-09-11:
It worked with a minor fix in 78ed984.
I still feel uneasy about it, like it will stop working the next time I touch something unrelated. Fingers closed.
Thanks a lot for your help and patience with me.