jotego/jtcores #54: Speed Music speed differences in Bionic Commando!

gomadosnitro, 2020-06-27:
I didn't test the game speed but it seems  slower on Mister but  i have to test it better. But the music is slower on Mister

https://www.youtube.com/watch?v=oN-ElnzxN4A&feature=youtu.be

Thanks.

jotego, 2020-09-09:
We were able to recompile Bionic Commando last week, so we can tackle this issue now.

**Can you measure the frame rate of your PCB? If you use OSSC you can see the frame rate in the LCD display. Or maybe your monitor shows Hz information on screen.**

I wonder if the PCB crystal has drifted during these decades. A drift is likely to be less than 0.1%. For crystals built after 1992, I found a figure: 10ppm per year. That would be 0.03% overt 30 years. Still noticeable if you leave the PCB on for enough time but very little. Maybe crystals used in 80's arcade PCBs have aged worse. A 0.1% would be noticeable after having the game on for a few minutes.

jotego, 2021-01-31:
I think this was fixed in the latest release. It was due to a design "feature" of the Yamaha timers where a bit is not reset when the time value is loaded (in the original design). 
I will close the issue. Feel free to re-open it if you think it is not solved.

closed 2021-01-31 by jotego