jotego/jtcores #60: Commando Alternative

eubrunosilva, 2021-01-11:
[Mercenario (Commando bootleg).zip](https://github.com/jotego/jtbin/files/5721755/Mercenario.Commando.bootleg.zip)

I know you said you didnt support alternatives but this one sometimes works and sometimes don't... some memory problem?

Thanks

jotego, 2021-04-18:
It looks like the first byte is differently encrypted on the bootlegs:

https://github.com/mamedev/mame/blob/cb8e92c227a7eee96affbc32dba255079a0f146c/src/mame/drivers/commando.cpp#L737-L738

I will need to detect the ROM being loaded to enable or disable that little difference. Or it can also be done by adding an extra byte to the MRA.

It needs some work. Maybe someone will contribute the support for it.

jotego, 2021-04-18:
Ok, it was an easy thing to add. It's done in 0b3a516b. I will update jtbin in a few minutes. Please make a PR to JTBIN with the new MRA files.

closed 2021-04-18 by jotego