jotego/jtcores #61: 1943 - Glitchy graphics in demo loop and stage transitions

M-Walrus, 2021-01-31:
In the demo loop you can see sprites glitching on various parts of the screen.  Look below the title on the right side while the background is scrolling past, and below the large planes as well.

Graphics glitches also occur during stage transitions on the image that shows during the stats screen.  Look in the middle-right of the mid-stage graphic (when in vertical/tate orientation) to see a line or square of glitched pixels.  This bug looks different on each of the transition screens, and is not visible on some of them (ones with the blue sky backgrounds.)

There also appears to be a small blue line next to the blue plane graphic on the title screen, before and after the demo loop.

Happens on all versions of the game. All mra's tested.

Example of demo loop graphics glitches:
https://youtu.be/MJazHnoimbY?t=20

Level transition screen graphics glitches:
https://youtu.be/MJazHnoimbY?t=249

Other examples timestamped in description of video.

Tested with most recent core as of  Feb 1st 2021

jotego, 2021-04-18:
-The problem is probably due to the map/scr reads
-Tried multibank access on cbe66a5d, but it didn't help
-Need to add cache for the map reads

jotego, 2021-05-19:
The blue line is in the original graphics. It might be a bad ROM dump. It can be corrected by editing the ROM file.

[image]

The other problem is solved now. I will update the RBF files for download later.

closed 2021-05-19 by jotego