jotego/jtcores #66: Street Fighter graphics issues

M-Walrus, 2021-05-01:
Current April 2021 build has the following issues (youtube links are timestamped to issue occurence)

None of this is confirmed to real hardware.

1. Guys on right side holding planks in the bonus stages- the bottom one has a line from the background going through him, and there's a line missing from arms on top:
https://youtu.be/5TVZO4VB1yY?t=825

2. When plane flies south on the map between stages, it splits in two:
https://youtu.be/5TVZO4VB1yY?t=838

3. Brown line appears on right side of screen when continually fighting Sagat. Timestamped right before it shows up. Persists through continue screens into the next round and reappears randomly:
https://youtu.be/5TVZO4VB1yY?t=1045

jotego, 2021-05-19:
I have fixed the plane bug in b9fd12. I will upload the file later today. There's a good chance that the bonus stage is the same problem. Could you please check it?

M-Walrus, 2021-05-22:
I checked and the bonus stages still have the same issue. Plane is most definitely fixed though, nice work! 
Updating original post to reflect this.

jotego, 2022-02-03:
It is due to a misalignment of the object and scroll layers. This got fixed when adjusting the LHBL signal going into the object circuitry for issue #70. Fixed in 518600b

closed 2022-02-03 by jotego