jotego/jtcores #67: Speed Rumbler - Game freezes when entering final boss

skutis, 2021-05-09:
Speed rumbler freezes when entering final boss.
- This was without car, perhaps not important.
- There was some garbled graphics in the upper left corner just before it froze.
- The music kept playing although everything else was frozen.

[image]

https://user-images.githubusercontent.com/80739822/117584869-9fe2d180-b10f-11eb-987d-e1c2d8f9ce8d.MP4

jotego, 2022-02-04:
I think this is due to the CPU address changing while RnW signal is low. The SDRAM controller doesn't like that. I will try making some changes...

jotego, 2022-02-05:
Not only the CPU address, but the data may also change too in that CPU module:

[image]

The image comes from the unrelated proto1 project but shows how data for a 16-bit write operation changes while RnW and address is stable. This doesn't suite the SDRAM controller at all

jotego, 2022-02-07:
I think this is fixed in issue 60f94df. Although, I haven't tested it playing the full game. I will update the binaries in a few minutes.

closed 2022-02-07 by jotego