jotego/jtcores #68: Side arms graphics issues

paulb-nl, 2021-05-22:
- Horizontal resolution is only 376px instead of 384
- 1px black column on the left
- Char layer is offset 3px to the left compared to Scroll layer

JT:
[image]

[image]

Mame:
[image]

[image]

jotego, 2021-05-23:
I am following the schematics for the screen size. Maybe I misread the blanking length, I would need to check.

The char can be offset by adjusting a parameter in the video module, it's a trivial change but it's hard to get a reference. The parameter is  CHAR_OFFSET. There are similar parameters for other elements. Please have a go adjusting it. In games where graphic elements get mixed to make an image, it's easy to do. Usually sprites and characters get mixed at least when entering a high score.

paulb-nl, 2021-05-23:
I have tried CHAR_OFFSET with positive values but it just moved the Char layer further to the left. Is it ok to use negative values to move it to the right?

Adjusting SCR_OFFSET to move the scroll layer left instead doesn't seem to work. It stays at the same position. I also couldn't get the last 2 pixels on the right of the scroll layer to appear after changing the blanking to show 384 pixels.

jotego, 2021-05-23:
> I also couldn't get the last 2 pixels on the right of the scroll layer to appear after changing the blanking to show 384 pixels.
But 384 is what was there before, isn't it? Do you mean 376?

The original circuit is simulated in the folder cores/sarms/ver/timing. I attach an LXT waveform so you can check it. I based the core timing on this simulation.

CHAR_OFFSET accepts negative values. But some value can have unexpected consequences as they may interfere with the tilemap read process. In general, I think it behaves well, though.

I am surprised SCR_OFFSET didn't have an effect. I checked the files and it is used. Please make sure to pull the files again, I think the public branch was updated but I'm not sure. By default, I push to a private repository and only update the public when things are more or less stable. If you're going to be looking at it, I'll try to push to the public one more often. Nowadays, I use a branch called *week* as the work branch, as my release schedule seems to revolve about weeks.

I changed the way the scroll works in games with ROM maps recently. I actually haven't released an updated RBF for this game, though. 

The difficulty of getting the offsets right is that each layer has a different latency when processing the pixels. The OFFSET variables try to compensate for that so the input to the colour mixer is balanced. The mixer itself has a different delay, and the BLANK_DLY parameter handles that.

On top of things, this should work when the flip bit is set too. And I think that the MiSTer backend may be moving things by a pixel, or may not be responding to the pixel clock enable the same way as the MiST backend (much simpler) does.

Sometimes I wish I had a generic graphic ROM to align all layers more easily in a simulation. This still wouldn't work for all games, as some arcades seem to make up for hardware misalignment in the software; so you cannot assume that X=0 means the same for all layers.

[timing.zip](https://github.com/jotego/jt_gng/files/6527858/timing.zip)

paulb-nl, 2021-05-24:
> But 384 is what was there before, isn't it? Do you mean 376?

I think there is a misunderstanding. The core is 376 now. Mame is 384.

> The original circuit is simulated in the folder cores/sarms/ver/timing. I attach an LXT waveform so you can check it. I based the core timing on this simulation.

I checked the LXT waveform. It has 16 microseconds (128px) of Hblank and 48 microseconds of active video (384px).

> I am surprised SCR_OFFSET didn't have an effect. I checked the files and it is used. Please make sure to pull the files again, I think the public branch was updated but I'm not sure

I pulled the latest changes and tried to change SCR_OFFSET again but it doesn't have any effect with Side arms so I used a negative value for CHAR_OFFSET.

The extra black columns seem to be caused by this in jtcredits: https://github.com/paulb-nl/jtframe/commit/cfdce870f8d6711739b5887362a18eaea51dae97

Making the pixels black there is not needed so I removed it and it fixes the black columns in Side arms and Contra.

Here I have fixed the layers position and changed the width to 384px: https://github.com/paulb-nl/jt_gng/commit/7fd182cc28833f6d372ec1047b73538b338526e1

paulb-nl, 2021-05-28:
Everything fixed now with version 0527. Thanks jotego.

closed 2021-05-28 by paulb-nl