jotego/jtcores #81: Speed Rumbler audio pitch is wrong

The-Iron-Goat, 2022-03-22:
The sound in Speed Rumbler/Rush 'N Crash plays at the correct speed, but the pitch is wrong. Sounds like it's shifted down an octave or so. MAME sounds correct. Here are samples recorded from my PCB:

[https://www.mediafire.com/file/yj6ogmflc99bap1/Speed_Rumbler_PCB.rar/file](url)

*Volume levels may vary on some of these tracks

jotego, 2022-04-16:
MAME uses a fixed 240kHz interrut, which can be derived from the vertical counter like other CAPCOM games. But this does not seem right because:

1. During the introduction scene the music doesn't finish and is interrupted when the game starts
2. The Z80 controls the interrupts in the YM2203, the way the FPGA core is done, is by connecting the YM interrupts to the Z80 and it is the Z80 the one that is programming the tempo in the YM. It would be odd that the CPU is prepared to do that but a fixed interrupt is used instead

It is true that the music sounds too slow. The pitch is linked to the tempo in the YM chip. I can increase the pitch by connecting the YM chip to a faster clock -like other CAPCOM games do- but then the tempo becomes too fast.

I need to pull out the PCB and make some measurements for this.

jotego, 2022-04-16:
Board Measurements:

1. The Z80 interrupts comes from the YM
2. The interrupt rate is 250 Hz -not 240 like MAME

The Z80 and the FM chips share the same clock at 4MHz

jotego, 2022-04-19:
Fixed in 3f08a10.
The YM2203 timer was not getting reset correctly and when the right clock was applied the FM chip, the pitch was right but the tempo was too fast (~320Hz for the Z80 interrupt). Once the timer code was fixed by applying recent changes to JTOPL (https://github.com/jotego/jtopl/commit/e1ee02adf1959a38cf5e29b5859fbd3813c0d636) the tempo was right (250Hz).
Other games using the YM2203 in this repository have been quickly tested and did not have side effects after the fix.
I will update the binary files for Friday.

closed 2022-04-19 by jotego