jotego/jtcores #84: Speed Rumbler - Vertical tearing on left side of screen

Aquas77, 2022-05-07:
Occasionally during horizontal movement, screen tears occur vertically on the left side of the screen. I've also observed the Car Health Gauge flicker a bit during this, it seems to be related?

I see the problem outputting analog as well as HDMI.

I have provided a couple video examples of it happening a few times within a minute.

https://www.twitch.tv/videos/1475781295 - At beginning of video and 13 seconds in.
https://www.twitch.tv/videos/1475781296 - At 5, 22, and 32 seconds in.

jotego, 2022-05-08:
I think there's a good chance that this is the legit behaviour. It looks like the game is updating the scroll registers in the middle of the frame and that will have an immediate effect in the image rendering. This will be more likely to occur if the CPU is struggling to keep up with the action.

I can block the registers so the update is only taken in during the vertical blanking. This could be set as an option. We will check the PCB to see if that blocking existed, but I have not seen it in any other CAPCOM game of that era.

Another possibility would be that the CPU speed is wrong (I doubt it, but I'll double check). One more way of fixing it would be a turbo option for the CPU, like the one in Double Dragon.

jotego, 2023-11-12:
After experimenting with some screens, I think this is due to the screen or mister.ini settings, not the core itself.

closed 2023-11-12 by jotego