jotego/jtcores #87: Makaimura freeze, glitch and yellow flash

Lmn-San, 2022-06-11:
I tested it on rev A, C and G

when playing and do some continue in the later level the game freeze.
 also when choosing the option Block Yellow Flash in OSD to ON (no yellow flash during game) the game glitch:

at the beginning of the first part of level 3 (in the cave):
 I had the text "Player One Ready" appearing a milli second in the middle of the screen during the play in the cave.
at the second part of level 3 (moving medama platforms):
 the background tiles get messed up in a vertical line black box appears.
then after a few attempt of death and retry (credit feed) the game freeze.

when choosing the option Block Yellow Flash to OFF (yellow flash during game)
I also had a freeze once at the level 6 after several credit feed.

also after watching the live play this week from Mikado playing the rev G (I believe it is this one, but in Japan they call it new version) the yellow flash are more present than in MiSTer.
it should appear at the begining of level 3 but I never had it once there.
https://www.youtube.com/watch?v=hlUQKIsLlHA&t=516s

jotego, 2022-06-12:
I think this erratic behavior is due to either a bad compilation (timing errors) or a problem with the SDRAM. 

I take care not to upload bad compilations but I will double check it.

Lmn-San, 2022-06-14:
Ok after some speed testing of my sdram, it seems that my sdram is not good. 
I ordered a new sdram module and I will see then.

Lmn-San, 2022-06-19:
Received my new SDram and effectively everything is working fine :)
Thank you for your help.

closed 2022-06-19 by Lmn-San