jotego/jtcores #88: Speed Rumbler - Alt MRAs do not have properly balanced sound

Aquas77, 2022-07-03:
Currently (Set 1), the default MRA for Speed Rumbler is the only MRA that has proper sound balancing. The sound effects are really quiet on the alts.

jotego, 2023-11-15:
Confirmed. The sets below have loud FM and quiet PSG channels. This is quite unusual as it shows either a hardware or software change in the main version of the game

- rushcrsh
- srumbler2
- srumbler3

jotego, 2023-11-15:
This requires a bit of messing with the files for technical reasons:

- the core does not use mem.yaml, so adding a header to the MRA is a bit messier -but no big deal-
- the core sound comes from module _jtgng_sound_, which provides fixed gain settings at the moment and it is shared by other CAPCOM-compatible cores
- it may be worth doing #402 first

jotego, 2024-01-20:
Fixed in #88

closed 2024-01-20 by jotego