jotego/jtcores #96: the speed rumbler (patch): compilation error

jtPerceval, 2022-10-28:
commit: cec1d65

Using command `jtcore rumble -q -p` I get this error

```
Error (12002): Port "sdram_req" does not exist in macrofunction "u_bank3" File: /home/samu/git/jtgng/cores/rumble/hdl/jtrumble_sdram.v Line: 296
Error (12002): Port "sdram_req" does not exist in macrofunction "u_bank2" File: /home/samu/git/jtgng/cores/rumble/hdl/jtrumble_sdram.v Line: 274
Error (12002): Port "sdram_req" does not exist in macrofunction "u_bank1" File: /home/samu/git/jtgng/cores/rumble/hdl/jtrumble_sdram.v Line: 242
Error (12002): Port "sdram_req" does not exist in macrofunction "u_bank0" File: /home/samu/git/jtgng/cores/rumble/hdl/jtrumble_sdram.v Line: 220

Error (12002): Port "ba0_dsn" does not exist in macrofunction "u_game" File: /home/samu/git/jtgng/modules/jtframe/target/mist/jtframe_mist_top.sv Line: 582
Error (12002): Port "ba1_din" does not exist in macrofunction "u_game" File: /home/samu/git/jtgng/modules/jtframe/target/mist/jtframe_mist_top.sv Line: 582
Error (12002): Port "ba1_dsn" does not exist in macrofunction "u_game" File: /home/samu/git/jtgng/modules/jtframe/target/mist/jtframe_mist_top.sv Line: 582
Error (12002): Port "ba2_din" does not exist in macrofunction "u_game" File: /home/samu/git/jtgng/modules/jtframe/target/mist/jtframe_mist_top.sv Line: 582
Error (12002): Port "ba2_dsn" does not exist in macrofunction "u_game" File: /home/samu/git/jtgng/modules/jtframe/target/mist/jtframe_mist_top.sv Line: 582
Error (12002): Port "ba3_din" does not exist in macrofunction "u_game" File: /home/samu/git/jtgng/modules/jtframe/target/mist/jtframe_mist_top.sv Line: 582
Error (12002): Port "ba3_dsn" does not exist in macrofunction "u_game" File: /home/samu/git/jtgng/modules/jtframe/target/mist/jtframe_mist_top.sv Line: 582

```

jotego, 2022-11-21:
Try fixing it by using an [include ](https://github.com/jotego/jtframe/blob/wip/hdl/inc/jtframe_game_ports.inc)for the game ports.

jotego, 2022-11-23:
fixed in e952247

closed 2022-11-23 by jotego