jotego/jtcores #136: mikie: unused filters

jotego, 2023-02-05:
The filter-control signals are ignored. It isn't clear whether these are mute signals or a change in the bandwidth of a LPF

https://github.com/jotego/jtkicker/blob/a935e94262a01129a1d61d6f8c446c7acc3709cd/cores/mikie/hdl/jtmikie_snd.v#L78

jotego, 2023-03-04:
@etriana85 unsolder one of the Konami switches and measure it on a protoboard for characterization. 
I think it is an analog switch, so there will be two input/output pins and one control pin. The control pin will set the impedance across the other two pins.
- When the control is low, the impedance (resistance) will be high
- When the control is high, the impedance will be low
Or vice versa, it may be that when the control is high, the impedance is high.

Tasks
- [ ] Unsolder a device from one of the Konami boards that have it
- [ ] Apply 5V and GND to the same pins as in the original board
- [ ] Apply 0V to the control pin and measure the resistance across
- [ ] Repeat with 5V applied to the control pin
- [ ] Resolder the device to the original board. Use a socket if possible

etriana85, 2023-03-10:
[image]
[image]

etriana85, 2023-03-10:
[image]
[image]

etriana85, 2023-03-10:
[image]
[image]

closed 2023-03-31 by etriana85

jotego, 2024-01-30:
the switch is on when the control input is high
[image]