jotego/jtcores #139: roc: Allow Flip Screen option for analog output on vertical games (Roc'n Rope)

capitaineflam25, 2023-02-05:
Thanks for this great core.

Sometime your vertical CRT is not orientated the right way and can't be physically rotated, a "Flip Screen" option would be a nice enhancement, especially for Roc'n Rope

jotego, 2023-02-26:
I have looked into it. The game handles the background screen flip in hardware but the sprite screen flip is done in software (when the _cabinet_ setting is _cocktail_). This means that messing with the hardware flip wire will only flip the backgrounds, not the sprites.

I don't think this can be done without significantly altering the machine hardware so I am discarding your request. Sorry.

closed 2023-02-26 by jotego

capitaineflam25, 2023-02-26:
Hi understand. Thanks a lot for the analysis.

Saroumane, 2023-10-12:
Hello, is it also impossible to flip background + sprites for games like "Sonic Boom" using `JTS16B 1.2.46` ?

jotego, 2023-10-13:
System 16 games can be flipped on MiSTer through the OSD at the cost of introducing one frame lag. 

The option is still not available for the Pocket. But it will. 

On other systems, like MiST, it cannot be done.

Saroumane, 2023-10-13:
Sorry, I forgot to say that I use a CRT, as the opener of this issue. Usually it needs an "hardware flip" through DIP switches. (I may not use the correct wording, but you'll understand what I mean).
I guess that most manufacturers of vertical arcade games implement it on their PCB, and you can re-create it in the FPGA core. 
But sometimes they don't (or do it halfway, just for the background ?). Is it the case for this core ?

jotego, 2023-10-13:
For CRT output, it is not always possible to add it. As my first answer says, for Roc'n Rope it is too messy to try to add it so I don't plan to do it myself.

Saroumane, 2023-10-13:
Thanks for the quick answer.
Is Roc'n'Rope running on the same core than 'Sonic Boom' ?
I have an idea (probably a bad one, but let's try it !) : what about using your patreon to make a survey about this ? And maybe if enough people interested create a target goal/tier to fund it ? Or a 'bounty' outside of patreon if it's possible ?
Just my 2 cents. (I perfectly understand that you don't enough time and manpower for these things)

jotego, 2023-10-13:
Roc'n Rope has its own core. The code is open source so anyone can add the feature.