jotego/jtcores #151: CPS1: The King of Dragons, cymbal PCM seems very loud.

Left-Empty, 2023-02-07:
Greetings,

In the newest RBF (April 10th's), the cymbal rhythm sound effect in the King of Dragons seems disproportionally loud.
It is very apparent in one of the boss fight's music (which luckily happens to be the first and second bosses'), but in some other parts as well.
Other rhythm effect used in the soundtrack don't seem to have that problem.

Thank you for your amazing work and best regards,

jotego, 2023-02-25:
Could you try modifying the "FX volume" setting in the OSD menu while playing the track? You can go into the test menu to play the songs directly. Let me know which one of the settings sounds right to you.

At the moment the settings go like:
very low  | low | high | very high
--------------|------|----|---
-9dB | -3dB | 0dB | +3dB

*high* is the default. I can shift all values down to make the default sound better and still give options to users.

@Left-Empty @TesterNes

Left-Empty, 2023-02-26:
Thank you for still working on it!

I'd say the ideal setting would be something just below low ( -4 or -5db?). But I don't have any real PCB to compare to, so I'm probably far from the most qualified to comment on this.
The issue remains on Stage 3 with the cymbals. That sample is being played many many times, and cut over before it completes any time, resulting in a very messy music track. It doesn't sound like that on the original game. 
To make a weird textual schematics attempt, the core sounds like “cyn-cyn-cyn-cyn-cyn-cyn-cyn-cyn-cyn"(slight pause and repeat) when it should sound "dru-dru-dru-dru--dru-cymbal"(pause).

TesterNes, 2023-02-28:
Hi Jotego, the overall volume of the fx seems to be fine at "high"(0db). Maybe the voice of the warrior when shouting I notice it a little low and the percussion (snare and bass drum) a little high.
Apart from that, the main problems are 3:

1- The frequency of some oki adpcm instruments is wrong:
    as you can hear below compared to the pcb, the entire set of drums and cymbals
    sound more compressed, as if they had less khz even without applying any filter on the sound
    ("original filter": off).
    Stage 1. code 0B.
https://user-images.githubusercontent.com/28839295/221765491-0fadb3ea-7ce2-4257-80c6-9210f93505cb.mp4
    Ending. code 12.
https://user-images.githubusercontent.com/28839295/221765760-8f56ef26-f92d-45a1-b298-e3202883fbcf.mp4

2- Bad timing in timpani/drum rolls: on the title screen and character introduction it's very common for
    the rolls have a different attack from the original, as well as mismatching some sections or skipping
    some blow. or maybe it seems as if they exchanged the bass drum for the timpani
    Title theme. code 17.
https://user-images.githubusercontent.com/28839295/221766304-41af9c61-33d0-4334-9b45-fb76b8f69363.mp4

3- The cymbals with the hi hat and ride function inside the oki channel have an excessive volume compared to the
     rest of instruments of the same adpcm channel and the other opm channels.
     Stage 4. code 04.
https://user-images.githubusercontent.com/28839295/221766663-17b523c3-8102-4f95-a16b-c49b5e53339d.mp4

Hope you can fix it.

jotego, 2024-01-20:
Completed on 612104bd

Left-Empty, 2024-01-26:
Thank you for the fix, it's... almost there ^~^;

Now I can hear a slight "pop" at the beginning of about every samples here sadly. It can be better isolated by turning on a low pass audio filter.
That pop could be heard in all the CPS1 games I gave a try. It seems to be dependent on the sample. For instant, in the King of Dragon, you can hear it very distinctly landing any jump, and in the music. Swinging the sword doesn't produce it though. 
Unrelated, but I tried turning off the FM or PCM and adjusting the sensitivity, but none of the settings seemed to do anything.

Cheers!

jotego, 2024-01-26:
Would you mind creating a separate issue for that report, please? I think it is unrelated.

Left-Empty, 2024-01-26:
I wasn't sure about that, I will do it!

h1deous, 2024-01-26:
I can confirm the popping, also the popping wasn't present on the previous version.

Left-Empty, 2024-02-01:
Reopening, as it is still an issue, and for the various references that were provided by @TesterNes.

jotego, 2024-02-01:
Could you check if the SNES version of the game sounds good? It can be easier to balance the instruments using that as a reference rather than PCB recordings. PCB recordings also have an uncontrolled sound signal chain which make them less reliable.

Left-Empty, 2024-02-01:
Thank you for reopening!

I just checked it, and in all tracks, especially that Stage 4 one, the cymbals volume is really nicely balanced on the SNES version.
Sadly, this nice effort of the port goes to waste because the HYPERLOUD DRUMS cover everything 😄 
On the SNES stage 4 track, the cymbals are barely heard in the background the majority of the time - the sample is being played quite relentlessly on both the arcade and SNES versions of the track, to create a sort of subdued rim effect in the background by using only the very first part of the sample. It might be hard to hear because of the aforementionned very loud drums.
The Stage 1 boss and other tracks sounded nicely balanced as well, excepting for, again, the ever present loud drums.

TesterNes, 2024-02-01:
I can contact with the owner of the pcb to make some recordings in service/test mode if that would help. The snes sound's quite different in comparison. It seems that sound is always very problematic to recreate in any kind of emulation... pity. Mame in this particular case is quite more accurate though.

jotego, 2024-02-01:
If you can get recordings with the same volume and settings for voices, sound effects and music, that would be great. In the end there are two parameters here to adjust:

- the balance of FM to ADPCM volume
- the cutoff frequency of the on-board low-pass filter (MAME does not implement this one)

TesterNes, 2024-02-18:
[King of Dragons (J) (910805) part1.zip](https://github.com/jotego/jtcores/files/14324455/King.of.Dragons.J.910805.part1.zip)
[King of Dragons (J) (910805) part2.zip](https://github.com/jotego/jtcores/files/14324456/King.of.Dragons.J.910805.part2.zip)

I got this wonderful recordings from https://www.youtube.com/channel/UC_zQPxykZVCd9TXxqT4DhIA He personally records them for us. 
As you can hear in code0b (Stage 1) at 0:14, the drums sounds less compress than mister's. 
And in code04 (Stage 4), cymbals volume is barely present. 

Thank you and thanks to VideoGamesCompleted for the recordings!

jotego, 2024-02-19:
Thank you. With the help of Artemio, I am running a deep comparison of the sound signal path of the core vs the real PCB. See #556

jotego, 2024-02-21:
This is a first adjustment based on the measurements Artemio did. Let me know how it feels, please
[jtcps1.rbf.zip](https://github.com/jotego/jtcores/files/14355727/jtcps1.rbf.zip)

Left-Empty, 2024-02-21:
As a really simple amateur this test build feels like a stepback to me: the whole rhythms track seem to be louder, especially the drums, and the cymbal is just part of the overall rhythm track taking over the music score. 
I am unable to tell if the cymbal track specifically is better balanced compared to the drums, as I don't have the musical ear to tell that — I am sure TesterNes will step in with a much more helpful comment!

Also quite a loud pop can be heard here when a track ends (ie, before the boss appears or at the end of a stage).

I am sorry I can't be more positive, but know my heart is in the right place supporting you Jose!

jotego, 2024-02-21:
Thank you for testing it. I was kind of expecting that. We found that ADPCM was relatively too damped compared to the PCB measurements. I had to amplify it by +2dB. The interpolation filter in the core, prior to any other filter, was upsampling the PCM data and removing aliasing from it. The PCB tries to remove aliasing but is much less effective. So by changing the antialias filter for something much closer to the PCB, the PCM volume goes up again at high frequencies.

Maybe this CPS board has different audio filters and gains than the other boards.

TesterNes, 2024-02-23:
I second everything said by @Left-Empty .Taking code0b (Stage 1) as reference from 00:14 to 00:18 is a good part to compare (ADPCM solo) and I can hear the sound is very muffled and compress. Maybe too much low pass filter? and there is something strange, maybe related to the aliasing: at the beginning of the drum roll from solo section, can be heared a fadeout not present on pcb.

Left-Empty, 2024-02-23:
Oh, what a cool toy you prepped for us, thank you :D

With or without any LPF set in the Mister, I'd say 11001100 (CC) sounds very close to what I'd expect, with both the drums being hearable and the cymbals just extending them and not covering them.
But, there is a but! At that setting, the samples used during the gameplay for the hits and such sound very low and/or quite muffled in balance to the music.

Can't wait what others may come up with, maybe with ideal settings.

Cheers!

Left-Empty, 2024-02-23:
Oh, what a cool toy you prepped for us, thank you :D

With or without any LPF set in the Mister, I'd say 1101100 (D8) sounds very close to what I'd expect, with both the drums being hearable and the cymbals just extending them and not covering them.
But, there is a but! At that setting, the samples used during the gameplay for the hits and such sound very low and/or quite muffled in balance to the music.

Can't wait what others may come up with, maybe with ideal settings.

For Knights of the Round, 1101100 (D8) works very well to my ears too, although I haven't played the game in over 20 years on an arcade machine.

Cheers!

(I preferred to repost this because I edited it so much and made mistakes, it would create confusion for people following the email thread, sorry for the slight spam)

jotego, 2024-02-24:
Thank you. Please try again with this filter that should not have the problem during gameplay. The bits have the same meaning as before. I recommend you start with the $98 setting.

[jtcps1.rbf.zip](https://github.com/jotego/jtcores/files/14392041/jtcps1.rbf.zip)

TesterNes, 2024-02-24:
00110110 (36) This value is the one I find most similar to the PCB on The King of Dragons. There are instruments that need to be adjusted within the ADPCM channel, hi-hat much, much lower and crash a bit higher. Lastly, I continue to report a rare sudden random drop in volume in the snare drum. If you manage to build a rbf with instruments volume like you did with gain and filter, I'll adjust it gladly.

jotego, 2024-03-01:
I think the problem with the previous one was in the filter implementation, which used a technique (IIR) that may produce ringing under some conditions. I have used a more conservative implementation (FIR) this time.

This version of the core does not allow for code manipulation. The values are fixed. This is the release candidate. Let me know how you like it.

[jtcps1.rbf.zip](https://github.com/jotego/jtcores/files/14460046/jtcps1.rbf.zip)

Left-Empty, 2024-03-01:
Sorry, I had forgotten to give a try to the previous build, but in this latest one, everything sounds awesome here... excepting for the insane cymbals, which are still extremely loud, to the point of covering everything in a couple of tracks.
The issue with cymbals is also present in Knights of the Round, but much less noticeable (the cymbal sounds a little harsh, but not as aggressive as in The King of Dragons).

TesterNes, 2024-03-02:
I still find the ADPCM sound a bit muffled compared to the PCB recordings, and the more I listen to the recordings and compare them, the more I realize how unbalanced the volume of some instruments is. Anyway, what worries me the most is that still, very randomly, there is a sudden loss of volume in the snare drum and as I have already reported in this thread, the roll of the title screen song is heard erratically at random, as if the attack of the drum set was inverted... very strange. Maybe a timing issue.

jotego, 2024-03-03:
The two known PCM filter responses are matched in the 22b08c09 core version. [jtcps1.rbf.zip](https://github.com/jotego/jtcores/files/14474490/jtcps1.rbf.zip). You can select the filter in the DIP settings menu. You can also adjust a bit the PCM volume in the OSD. I think this is the best we can do so I am closing the issue again.

[image]
[image]

closed 2024-03-03 by jotego

TesterNes, 2024-03-07:
The main theme of this issue was the cymbal volume inside the oki channel, not the filter of the adpcm/pcm. It is not directly related.

jotego, 2024-03-07:
The volume for the PCM channels is correct and verified. If it feels different, it can only be because of filtering along the signal path. 
This is the best we can do without having that specific board, which might be different to other CPS1 boards.