jotego/jtcores #157: CPS1 Carrier Air Wing sprite priority bug

PhantombrainM, 2023-02-07:
Yesterday I played Carrier Air Wing all 9 (or 10?) rounds with Continue dip on.

I noticed when the plane is fueled up (all three rounds or so?) that the pipes are flying through the air and I thought it looks weird.

So I opened Youtube and found the same behaviour but also some YT Vidoes without the problem. I think it's a bug.

Wrong (like on MISTer):
https://www.youtube.com/watch?v=ev1rgJm6uWw&t=545s

Seems right:
https://www.youtube.com/watch?v=T2ffq0MHzGI&t=306s

Also I lost control during the game at stage 7 or 8 (no control anymore and I died). Then I noticed I need to press start again and from that on I was player two (?) until the end of the game. Strange. I don't know why this happened and if it was a one time issue or not. Maybe I'll play it again tonight.

asturur, 2023-09-14:
Just to test i tried this game on the latest cps1 compiled core and nothing changed. I heard that there was some DTACK changes in the cpu of the core and i wonder if it could make any difference. it didn't.

asturur, 2024-05-13:
I tried the latest cps1 compilation from the github actions, the one with the fix for the dtack and cen, from this run https://github.com/jotego/jtcores/actions/runs/9055203490. The issue still persist.
Not that there was mention that it could have been fixed, but i guessed a little test was ok to do.

paulb-nl, 2024-09-12:
> The commit 0fdcb02 that fixed this issue, broke
> "Three wonders" intro. See jotego/jtcps#131
> Final Fight and Varth alignment. See jotego/jtcps#132

I am working on this issue. The jtcps repo is gone so I can't see the things that broke with your previous fix. What potential issues should I look at?

Now my changes fix this hoses issue and the Nemo heads animation is only inside the letters like Mame and 3wonders shows the gun in frame only like the pcb footage.

asturur, 2024-09-12:
I remember final fight had a stage where there was an elevator and the layers were messed up

jotego, 2024-09-13:
> I am working on this issue. The jtcps repo is gone so I can't see the things that broke with your previous fix. What potential issues should I look at?
>
> Now my changes fix this hoses issue and the [...]

I have made the original repo visible again, so the links work now. I will hide it again eventually because it creates confusion now that the source code for all cores is centralized in a single repo.

Thank you very much for looking at this. I am looking forward to your patch.

paulb-nl, 2024-09-14:
Thanks Jotego. I sent the pull request with my changes. 

I did not see any layer alignment issues during my testing.

jotego, 2024-09-15:
Fixed in #802 thanks to @paulb-nl. It introduces a regression in Nemo's animation, though.

closed 2024-09-15 by jotego