jotego/jtcores #165: shanon: band in the middle of the road

jotego, 2023-02-07:
commit: 831a6bc

There is a transparent band in the middle of the road. It looks like it should overlap with the road markings and not be transparent.

[image]

jotego, 2023-09-17:
The problem occurs because the _cent_ signal (centre) is not aligned with the road bitmap centre. Thus, the road layer type (sky, ground, etc.) signal _rrc_ carries the wrong information around the centre of the road

[image]

closed 2023-09-17 by jotego

asturur, 2023-09-17:
will this fix also outrun similar lines? or cores are different enough for that not to be the case?

jotego, 2023-09-17:
Yes, it fixes both cores.

asturur, 2023-09-19:
tried this and looks great, outrun finally without the line in the middle is great.

commit e610b810 (2023-09-17): Fixes #165: shanon/outrun band in the middle of the road