jotego/jtcores #167: Golden axe buzzing corrupt sound in attract screen

shuffle5, 2023-02-07:
Happens on the attract sequences. Seems to affect Tyris Flare,in the village with Gillius & when she's with Ax Battler by the Eagle (moving eye graphic). You can't miss the annoying high volume buzz noise. 
Im using US Set 5

jotego, 2023-06-26:
Confirmed that the noise comes from the FM module. Is the Z80 sending bad parameters, or as these parameters triggering noise in JT51?

- [ ] Create VGM file with mamevgm to isolate the Z80 from the problem and resimulate in JT51 alone

jotego, 2023-07-03:
This occurs in the same PCB type as issues #294 and #172

jotego, 2023-07-03:
That board type has three devices different from the rest: a multiplier, a timer, and an arbitrarily added second t
- timer2 can be removed in MAME without causing this issue
- without the multiplier, there are some GFX problems in MAME but the issue does not happen
- without the timers, collision detection does not work

jotego, 2023-07-03:
Fixed with #294

closed 2023-07-03 by jotego