jotego/jtcores #168: Outrun acceleration and gear shifting are not as expected

PistolsAtDawn, 2023-02-07:


jotego, 2023-03-25:
I have tested these claims again in the public version. And at least I can say that:

- It is **not** possible to reach max speed without being in high gear
- Shifting down to low gear while at top speed __does__ cause you to slow down suddenly

In any case, this sounds more like a software configuration (DIP swithces?) or controller issue and not related to the arcade hardware (core) itself. So I am closing this issue. Thanks for reporting it.

closed 2023-03-25 by jotego

PistolsAtDawn, 2023-03-25:
Well, all I can say is that I am not able to replicate that behavior. On MAME and the Nintendo 3DS Arcade Classics release (not homebrew) it is not possible to exceed 194 km/h in low gear, and if you shift from high gear to low gear, you will slow down to 194 km/r.

To test this thoroughly against my MiSTer, I setup a fresh installation using Mr. Fusion and update_all.sh. The ROM of game pulled by update_all has a md5 checksum of:

/media/fat/games/mame# md5sum outrun.zip
e731f4792e481349b8aed3510cb3feab  outrun.zip

This matches the copy I have on my Steam Deck.

On this fresh MiSTer installation, I am able to start from 0 and hit top speed (well in excess of 194 km/r) without changing gears. At this top speed changing gear makes the tires spin, but there is no change in speed.

Are we perhaps not using the same ROM?

The graphics look beautiful with this release, by the way. I appreciate the effort that went into that.

ksdomino, 2023-03-26:
Same as what Jotego says: Works perfectly here too (rom from update_all). Staying at or below 194 in low gear. Also changing gears from high to low drops speed down to low gear max speed of 194, video proof: https://www.dropbox.com/s/y86v2yo6cb0kxcr/20230326_085038.mp4?dl=0

PistolsAtDawn, 2023-03-26:
> 

I see and acknowledge that it is working for you. I simply cannot get that same result, and I am having no luck determining why. 

Can you please help me determine why I cannot get that to work for me on a brand new fresh installation of MiSTer updated with update_all?

What is the md5 checksum of your ROM? Mine (pulled by update_all) is:
`/media/fat/games/mame# md5sum outrun.zip
e731f4792e481349b8aed3510cb3feab outrun.zip`

What is the md5 checksum of your jtoutrun.rbf file? The one uploaded yesterday has a value of:
`10e765b74dd939cc99c6bac215b593f2  jtoutrun.rbf`

The version uploaded Friday has a value of:
`a2296d2c221924d9669e6ffd52d5d95e  jtoutrun.rbf`

The one from yesterday (10e765b74dd939cc99c6bac215b593f2) is missing the second page of options in the MiSTer OSD. Other users on the forum report this same issue. This is the version that update_all is currently grabbing.

I have no custom settings set whatsoever, and I have even cleared the RAM within the service menu of the game and cleared the core settings through the MiSTer OSD.

Is there anything else you can think of?

Thank you for your help!

ksdomino, 2023-03-26:
Check your DM on discord, I sent you the files I'm using so you can compare, test and check the hash if needed if it still doesn't work..

PistolsAtDawn, 2023-03-26:
Thanks to ksdomino, we have this behavior figured out: I was not aware (and am not the only user) that the shift button needs to be held down in order to stay in low gear. Using gamepads, this is not very intuitive as only one button is set for the gear shift just like in MAME.

The acceleration issues, etc. I reported are not an issue as long as I remain in low gear by holding down the shift button on my gamepad.

Would it be possible to change this so that either two buttons can be used for high/low gear or to make it match MAME's behavior and have it be a toggle setting?

Thanks!

ksdomino, 2023-03-26:
Seems like a good idea, as not everyone has a wheel..maybe it's possible to add an option called "Gear Latch" on/off? So those of us with gear systems or cabinets can leave it off (this should be the default as it is the current behaviour) and people with pads can turn it on so the gear latches on (low gear) until they press the gear button again to unlatch it (high gear)..

jotego, 2023-03-27:
I'm glad this was just a misunderstanding. Let me get the remaining graphic glitches fixed before adding the gear option.