jotego/jtcores #174: Body Slam (World, S16) - not booting if the MCU is enabled

sqwirral, 2023-02-07:
`jts16_20211105.rbf` works fine, but I couldn't get any builds newer that to load the game.

Body Slam (World, S16) [8751 317-0015].mra

jotego, 2024-01-22:
The MCU was not sampling the data from the main bus correctly. Yet fixing it in 8f91537 made no difference.

gyurco, 2024-01-24:
In Body Slam, there's a bus request-bus grant cycle, after that the MCU resets the CPU and never releases the reset. So it might read something scary during bus arbitration?

gyurco, 2024-01-24:
Looks like it's not deterministic, it can reach CPU reset in various ways. Maybe because of the non-cycle exact MCU.

jotego, 2024-01-24:
One thing about the SEGA System 16 is that RAM had a battery so the game usually expects the RAM to have some valid content. It can initialize it too, but you often find that it reacts to old data. That might be affecting it. Maybe it is deterministic after a power cycle (power off and on).

It might be possible to wipeout the RAM (held in the SDRAM) by modifying the game load process, but I wouldn't go into that rabbit hole just yet. Probably a power cycle will do.

The MCU is meant to get access to the main CPU bus and probably check stuff, particularly check that the ROM has not been tampered with. That was my main focus when debugging it.

gyurco, 2024-01-24:
Strangely If I reload the core via USB Blaster (clearing registers and probably BRAMs), then the game starts.

I observed the MCU does the coundown from 3 to 0 at the beginning (it can write to VRAM), but currently only 0 flashes for a moment after a reset, then it hangs.

gyurco, 2024-01-26:
Some observations:
- MCU to CPU reset is inverted (p1[6])
- MCU doesn't use bus arbitration for 68k bus access, but put the 68000 into reset while accessing RAMs/ROMs

gyurco, 2024-01-26:
- Something must be done with the PPI, too (maybe use cpu_rst), as during initialization, when the CPU mostly in reset, the video must be enabled. But Using the cpu_rst for PPI also activates flip, too, which is unneeded.

Implementing the 3 observations above makes the countdown 3-2-1-0 appear (just flipped). It still freezes after that, debugging continues...

jotego, 2024-01-26:
The reset signal can be a bit tricky too. I think it was in the JTS16B where a bug was fixed by adding support to the M68000 driving its own reset pin low. Not sure if that could play a role here too.

gyurco, 2024-01-26:
It would be strange if they allowed reset by both the 68K and the 8751, too. BTW, is there any System 16A schematics with the MCU out there to check if BR, BG and BGACK are really used?

gyurco, 2024-01-26:
After some changes, it looks like ROM check is passing now. Now RAM reads returns suspicious values.
I have a question about the work RAM handling: is RAM mirroring is in place? E.g. when the MCU reads 4200h, it should read the RAM at 200h, but it'll set A=c74200h, and
```
assign cpu_addr = A[12:1];
assign rom_addr = {1'b0, A[17:1]}; // only 256kB on System 16A
```
doesn't have[13:1], so which one is used to address work RAM?

If [12:1], then it's only half size, if [17:1], then mirroring is broken.

gyurco, 2024-01-26:
Well,
```
assign rom_addr = {1'b0, rom_cs ? A[17:14] : 4'd0, A[13:1]}; // only 256kB on System 16A
```
looks ok. But as you mentioned, it probably does an NVRAM check (the whole workram area is NVRAM).

gyurco, 2024-01-26:
Is there any easy method to clear the RAM area, or load it from the MRA/include it in the ROM?

gyurco, 2024-01-27:
> Is there any easy method to clear the RAM area, or load it from the MRA/include it in the ROM?

Seems it's not important. The game finally starts! However there are some issues left:
- The boot screen is flipped
- The game cannot be resetted (need to clear the MCU RAM, because the 8751's prog checks some bytes if already initialized, and goes berserk with continuously resetting the 68k).
- In-game timer doesn't count (maybe need to connect t0_i to intack, as on the schematics? VPA from MCU also doesn't work, probably because of that).

jotego, 2024-01-27:
Congratulations! Great news!

The RAM initialization problem could be fixed by changing the jtframe_ram in jtframe_8751mcu to a dual_ram. Then have one port driven by the prog signals and just write zeros to it while prom_we is toggling. 

The boot screen flipped may or may not be a real issue. Sometimes DIP settings are not read until some initialization is done so it is not unusual to have flipped screens. 

It is worth checking if there is some connection missing both related to the flip and to the timer.

jotego, 2024-01-27:
Also, please check that simulations behave as you see on the FPGA compilation too.

jotego, 2024-01-27:
FYI

Simulations have a powerful tool, the X/Z signals, which will tell you whatever is not initialized or connected. To see them you need to use a different simulator, like iverilog or modelsim. **jtsim** can use them, but they are terribly slow. It is ok to simulate one or two frames and then plot signals looking for red (X) and orange (Z) in gtkwave.

gyurco, 2024-01-27:
I just added a bunch of ram_prog signals to jtframe_8751mcu, so I can reset the RAM when mcu_rst triggers, works well (except that at reset, the VRAM is also not cleared, so the boot screen is cluttered).

Now I have a problem with simulation:
```
[cores/s16/ver (master)]$ jtsim -setname bodyslam -q -video 10 -w -d VERILATOR_KEEP_SDRAM
jtsim must be called from within the ver folder of the core
```
As it can be seen, I'm in the 'ver' folder.

jotego, 2024-01-27:
go into the game folder, or create a new one under ver:
`cores/corename/ver/game`

jotego, 2024-01-27:
> I just added a bunch of ram_prog signals to jtframe_8751mcu, so I can reset the RAM when mcu_rst triggers, works well (except that at reset, the VRAM is also not cleared, so the boot screen is cluttered).

For the VRAM and other M68000 memory, I want to do it via MRA, by loading blanks in it. It will work for swapping cores.

By the way, does the `load ROM` OSD option work for you? It stopped working on MiST a while ago. I don't know if something changed in the firmware and I just need to update. Sorry for the side question

gyurco, 2024-01-27:
```
By the way, does the load ROM OSD option work for you? It stopped working on MiST a while ago. I don't know if something changed in the firmware and I just need to update. Sorry for the side question
```
As it doesn't apply core_mod, I didn't bother with it. But there's a better trick to try any variant without Load ROM: if SW1 is activated (so preventing FPGA reprogramming), you can change to any other ARC file via load core, it won't reload the core, but apply everything else (mod, rom, switches, etc...)

jotego, 2024-01-27:
Keep an eye on the other games that have **8751** in the title. They should come alive too as you fix this one.

gyurco, 2024-01-27:
Ok, simulation runs now. As I see, the MCU rom wanted to be preloaded from mcu.bin, but it's not there. I hope the usual programming from .ROM file will work.

The only change I did in 8751 is the added ram programming signals...maybe Verilator will complain about unconnected pins, but shouldn't break anything else.

gyurco, 2024-01-27:
I get this in sim from frame 49:
[image]

I estimate the game will start at somewhere around a thousand frames, so I won't wait for it :)

jotego, 2024-01-27:
> I estimate the game will start at somewhere around a thousand frames, so I won't wait for it :)

LOL. It won't take long. Leave it running in the background without dumping signals. It's good reference to keep video folders of long sims so you know at which frame different things happen. S16A can be simulated without sound `jtsim -q` and run correctly. That speeds it up.

gyurco, 2024-01-27:
I leave it here for 10000 frames with -q -w.
Let's see what'll happen. But I don't think it'll differ from the FPGA :)

jotego, 2024-01-27:
The `-w` will dump the signals and 10,000 frames is probably a lot of GB in a test.lxt you may not have enough RAM to load it!

gyurco, 2024-01-27:
How can I prevent to dump the signals?

jotego, 2024-01-27:
By default, just `jtsim`, it won't dump anything. When you add `-w` it will dump them. If you add a number after `-w` then it will dump them after that frame `-w 50`

gyurco, 2024-01-27:
Ok, re-running without -w.
Frame234:
[image]

gyurco, 2024-01-27:
It was a bit earlier than 1000, frame 640:
[image]

jotego, 2024-01-27:
Great to see it after all this time...

gyurco, 2024-01-28:
Timer is fixed.
Ace Attacker and Quartet's controls are suspicious, but I guess it's not MCU related.
dumpmtmt's MRA is not created, why? (deleted from skip.setnames). Not a big problem, it's Japanese Body Slam.

jotego, 2024-01-28:
> Timer is fixed.

Love it! Thank you

> Ace Attacker and Quartet's controls are suspicious, but I guess it's not MCU related.

What does suspicious mean? MCUs often read the inputs and pass them to the main CPU. You can check whether this is the case with a breakpoint in MAME or by checking access by the MCU on signal tap/simulation

> dumpmtmt's MRA is not created, why? (deleted from skip.setnames). Not a big problem, it's Japanese Body Slam.

It must have been broken at some point. Maybe the timer? Games that work correctly can be taken out of the skip list.

gyurco, 2024-01-28:
According to MAME, Quartet has different input mapping than Quartet 2, and Ace Attacker has some custom input device, these has to be added I think. Not sure how Ace Attacker is supposed to work with keyboard.

I deleted dumpmtmt from the skip list, yet no MRA from jtframe mra. Where does it looks for MAME sources?

jotego, 2024-01-28:
> I deleted dumpmtmt from the skip list, yet no MRA from jtframe mra. Where does it looks for MAME sources?

Look for it in the `_alternatives` folder

gyurco, 2024-01-28:
> > I deleted dumpmtmt from the skip list, yet no MRA from jtframe mra. Where does it looks for MAME sources?
> 
> Look for it in the `_alternatives` folder

Bingo!

gyurco, 2024-01-28:
Now to implement Quartet's controls, what's their GAME_ID? macros.def don't have them.

jotego, 2024-01-28:
The header contains flags to enable different hardware features. The game_id is at offset $18

```
[header]
fill=0
data = [
    { offset=0x10, dev="fd1089a",      data="01" },
    { offset=0x10, dev="fd1089b",      data="02" },
    { offset=0x11, dev="fd1094",       data="01" },
    { offset=0x13, dev="i8751",        data="01" },
    { offset=0x18, machine="sdi",      data="19" },  # enable trackball
    { offset=0x18, setname="afighterb",data="1d" },  # enable analogue input
    { offset=0x18, setname="afighterd",data="1d" },  # enable analogue input
    { offset=0x18, machine="passsht",  data="02" },  # enable four players
    { offset=0x18, setname="quartet",  data="04" },  # enable four players
    { offset=0x18, setname="quarteta", data="04" }   # enable four players
]
```

So _quartet_ will be game_id=4. Note that S16A's macros.def is included by S16B too. Just add one more line for **GAME_QUARTET** in macros.def. Note that the macros are replicated as local parameters in the modules, basically to get rid of warnings (both from Quartus and simulators) related to bit width.

jotego, 2024-01-28:
By the way, s16b's TOML file defines many more values for offset $18. There is some encoding built in, so we should not change them lightly.

gyurco, 2024-01-28:
Luckily Quartet already has its number.

Now a feature request popped in my mind: trackball support via mouse would be nice on SDI :)

jotego, 2024-01-29:
It may already be working like that on MiSTer/Pocket. I think I have not paid attention to analog inputs on MiST/SiDi

gyurco, 2024-01-29:
Looks like only analog_joystick signals are connected, nothing mouse-like. BTW, I updated user_io which supports right stick on gamepads (at least on my cheap one from Tesco), it's much better to play SDI :)

However I noticed on stage 2, there are a lot of blank lines at the right edge of the screen when lot of objects attacking.

jotego, 2024-01-29:
> Looks like only analog_joystick signals are connected, nothing mouse-like. BTW, I updated user_io which supports right stick on gamepads (at least on my cheap one from Tesco), it's much better to play SDI :)

Nice! 
 
> However I noticed on stage 2, there are a lot of blank lines at the right edge of the screen when lot of objects attacking.

Yes, that's a bug I've seen. No one has complained about it though. I had tried to debug it once or twice with no luck

gyurco, 2024-01-29:
> Yes, that's a bug I've seen. No one has complained about it though. I had tried to debug it once or twice with no luck

As those bars are random, I assume SDRAM delay problem. Isn't it possible to move some video RAMs to BRAM in large FPGAs (and leave it in SDRAM for the small ones)?

gyurco, 2024-01-29:
Also looks like the scrolling is broken at specific places, probably a counter is overflowing at the wrong time.
I can look at these if you want.

closed 2024-01-30 by jotego

jotego, 2024-01-30:
If I had a dollar for each typo!  Thanks for spotting it

One side effect of the counter is that one core does not in SiDi anymore. It was that border line that a 7-bit counter made the difference! 
Did you need to add it for resets after power up? If you start the core and never reset it manually, this function is not needed, is it?

gyurco, 2024-01-30:
I don't think JTTORA required RAM reset at all. Body Slam works after core bootup, as BRAM is zeroed during programming, but then it never starts again after a reset (same in MAME with soft reset - probably it was a protection feautre? The MCU program explicitly checks some initialized values at start, and goes into an infinite loop if they're found. But does the arcade cabinet have a reset button at all?)

jotego, 2024-01-30:
No reset buttons at all. We do need the reset functionality for applying DIP switches without re-loading the core. So it's good to have it. Even if JTTORA does not need it, having it there may prevent us from having to fix this again in another core in the future.

gyurco, 2024-01-30:
> It was another core, JTTORA, I think

|jtframe_ram_rst:u_ramu|	1996 (8)	0
|jtframe_dual_ram_cen:u_ramu|	1988 (1988)	0

The main problem is that jtframe_ram_rst doesn't infer BRAM for some reason, eating a lot of registers (2k), even on other FPGAs. I think it's not very economical for a 128 byte small RAM.

gyurco, 2024-01-30:
Info (276007): RAM logic "jts16_game_sdram:u_game|jts16_game:u_game|jts16_main:u_main|jtframe_8751mcu:u_mcu|jtframe_dual_ram_cen:u_prom|mem" is uninferred due to asynchronous read logic

jotego, 2024-01-31:
Thanks for spotting it! I've changed the code to use multiplexers. Quartus is a bit whimsy about inference and it has not got any better since version 17 going to version 21 or 22. I already pushed the changes

gyurco, 2024-01-31:
I know, it's a heads and tails game sometimes.