jotego/jtcores #177: HB generation and length

jotego, 2023-02-14:
Pins 22 (HBLK) and 23 (SYNC) are connected together on Extermination and Arkanoid 2. But the connection is different on Extermination X. It looks like HB and SYNC can be wired together to produce some logic (wired _and_?) but we don't know exactly the signals that each pin would produce individually.

Pin 18 (UK) may be a test pin, but there is a chance it is related to the HB/VB/SYNC signals. It could select the sync polarity. Let's try forcing it to VSS and check if the signals change in a meaningful way.

We want to know these signals

- [ ] HB/SYNC output when wired together @etriana85 
- [ ] HB when isolated @etriana85 
- [ ] SYNC when isolated @etriana85 
- [ ] Repeat the measurements with pin 18 (UK) cut from the board and connected to VSS @etriana85 
- [ ] Document the results in the design files @jotego 
- [ ] Document the results in the schematics @skutis and @etriana85 

Include /VSYNC and /HBLK in the three measurements as reference.
@etriana85 unsolder the chip and solder a socket, so we can make the tests without breaking the original pins.

etriana85, 2023-03-06:
Measurements of the osc (12MHz) pcb exterm:
The output of the osc moves between 10 and 12.5MHz.
Video attached.
In the sch, the osc directly attacks the SETA X1-001, without going through any inverter. It has a pull-up connected, and a resistor that is not going anywhere.
The output CLK of x1-001 only attacks the input CLK of x1-002.
Attached image.
[recording-2023-03-06-10-32-38.webm](https://user-images.githubusercontent.com/92724765/223077271-c771dc2c-d843-4746-b919-8b012b4494f3.webm)
[image]
[image]

etriana85, 2023-03-06:
These are the measurements of the test CLK, mounted on the proto
[image]
board:
The output of the osc moves between 11.9 and 12.2 MHz.
I attach images and video.
[image]

etriana85, 2023-03-06:
[clk_test_exterm.webm](https://user-images.githubusercontent.com/92724765/223111715-85d08b2d-1c94-43f4-bacb-e85b73a8b6d8.webm)

etriana85, 2023-03-06:
Attached two videos with the requested measurements. The first with the original connection to VCC (Uk), the second with the connection to VSS. I see that the two signals (HBLK AND SYNC) are kept at 0 (zero), whether they are at VCC or VSS, together and separately.
These measurements were made on the pcb as is, without using the test CLK that was mounted on the proto board.
As such it seems that we have the osc in good condition, I made the measurements.

etriana85, 2023-03-06:
[VSS.webm.tar.gz](https://github.com/jotego/jtcores/files/10898319/VSS.webm.tar.gz)

etriana85, 2023-03-06:
[VCC.webm.tar.gz](https://github.com/jotego/jtcores/files/10898324/VCC.webm.tar.gz)
[VSS.webm.tar.gz](https://github.com/jotego/jtcores/files/10898325/VSS.webm.tar.gz)

jotego, 2023-04-03:
@etriana85 try again using the new signal generator.

- Be careful not to produce more than 5V with the generator
- Pass the generator output through an inverter first, then connect the inverter output to the DUT

etriana85, 2023-04-03:
I'm working on this.

etriana85, 2023-04-07:
Measures Seta X1-001:
CLK_IN.............12MHz
CLK_OUT..........11.07~12.01MHz
nVBLK...............59.2Hz
VBLK.................59.3Hz
nHBLK2............15.6KHz
nHBLK..............15.6KHz
nVSYNC............59.2Hz

Observatión: Pin 46 (HC0) does not correspond to the sequence of the HC[8..0] counter.

etriana85, 2023-04-07:
Information Seta X1-003:
The output signals (pin55) nHBLK (15.6KHz) and (pin56) nVSYNC (59.2 Hz) of the SetaX1-001, were simulated with the signal generator and an inverter. These were connected to the input pins of the SetaX01-003 (pin 19 - 20), respectively.
No logic signals are observed on the output pins (22-23) HBLK - SYNC, applying Jotego's instructions, given in this issue.
It is concluded with Jotego that the Chip SetaX1-001 is broken.
I share images of the test.
[image]
[image]

closed 2023-04-24 by jotego