jotego/jtcores #181: Kabuki Z beta

jotego, 2023-02-14:


jotego, 2024-02-09:
Kabuki requires three Z80 CPUs, currently the core is done with only two

gyurco, 2024-02-27:
Added inputs and palette RAM handling to KabukiZ, but seems even the graphics HW needs adjustment, both tilemaps and sprites are messed up.

gyurco, 2024-02-27:
Sound added to KabukiZ. For TNZS, it needs some adjusments, probably will require another flag in mame2mra.toml, but not a big deal.
Graphics however is garbage.

gyurco, 2024-02-27:
Another problem the core no longer fits for MiST/SiDi. Maybe Kabuki can be skipped for these or the big banked RAM could be moved into SDRAM.

jotego, 2024-02-28:
I tried compiling it. It's awesome that it's booting up, colors are ok, inputs work... very good progress.

I think we can skip this for MiST/SiDi with some precompiler setting, like the **NOMCU** used in other cores. I'd rather not compromise the core performance or complexity.

The graphics seem to have some bad address mapping in part of the demo. The schematics may give hints about that. The most worrying part are the sprite glitches, which may be something else. Hopefully, it only requires applying some bit transformation to the tile and sprite ROM addresses based on some bank register or similar.

gyurco, 2024-02-28:
What's the difference between 'setnames' and 'machines' in the .toml file? Both are used in the data= section.

gyurco, 2024-02-28:
'machines' has all 'setnames' for a particular game?

gyurco, 2024-02-28:
Good news is that TNZS new version works.

jotego, 2024-02-29:
> 'machines' has all 'setnames' for a particular game?

This is MAME's terminology. _setname_ is the name of a given set of ROM dumps used for a specific game. _machine_ refers to a hardware architecture. Each _setname_ must belong to a _machine_. The main _setname_ of a given _machine_ usually has the same name.

The core TOML file accepts _machines, machine, setnames, setname_ in many fields in order to select the ROM sets to which the field is applied.

> Good news is that TNZS new version works.

Brilliant! Thanks :-)

gyurco, 2024-02-29:
Ahh, I see, it's synced with lut_cen.

jotego, 2024-02-29:
> In MAME, there's spriteram copying, probably double-buffering, done during vblank. Missing that would explain the heavy flickering.

I think the core can access two different pages in the memory. I need to refresh my memory but I saw more evidence for the object look-up table being handled as different pages rather than as a shadow copy done with DMA. If I did not find evidence of the CPU being held during DMA or signals in the PCB, I must have gone for the page scheme. But maybe I am not getting the page from the right register.

I usually write a lot of things, so it might be noted in schematics, or issues, or the README files. I also sometimes edit the MAME files I copy with changes and clarifications. When it is worth it, I make a PR to mamedev about it.

gyurco, 2024-02-29:
Looks like DMA can be toggled with cfg[1][5]. Usually all cores set this to 1 (no DMA). Kabuki Z sets it to 0, which means one page is copied to the other during VBLANK. I did this, and the sprites are no longer flickering.

gyurco, 2024-02-29:
I still have no idea about the wrong tiles. The Kabuki Z's 4 ROM dumps cannot be shuffled around.

gyurco, 2024-03-01:
Tweaking DMA fixed most problems, the only remaining one is that tile codes are 14 bits, not 13.

jotego, 2024-03-01:
> Is it possible to run just the graphics HW in sim (with VRAM dumps)?

Yes, but it looks like you are already past this need. The PCM sound is very loud. I am trying this combination now:

```
    pcm_gain <= kabuki ? 8'h07 : kageki ? 8'h0A : 8'h0;
    fm_gain  <= kabuki ? 8'h40 : kageki ? 8'h20 : 8'h30;
    if( !fm_en ) fm_gain <= 0;
```

gyurco, 2024-03-01:
> > Is it possible to run just the graphics HW in sim (with VRAM dumps)?
> 
> Yes, but it looks like you are already past this need. The PCM sound is very loud. I am trying this combination now:

It wouldn't help anyway, as problems were caused by race between CPU/DMA. Looks like it specifically enables to run tilemap DMA when it finished moving data in the VRAM.

Audio volume is always problematic...

jotego, 2024-03-01:
It is looking quite good indeed. Thanks a lot.

The first boss is still blinking. I think there is something missing. The CPU must either be held during DMA transfer or needs a way to learn when DMA is over. If look at other cores, a _dma_busy_ signal is often fed back to the CPU. Maybe MAME can get away with it because DMA does not consume emulation time, so it does not affect the CPU.

That is another problem if the current implementation. It is too fast. It should run at one of the available clocks on the board. Look at the schematics for hints about which clock might be used. I have gated it with pxl_cen (6MHz) for now.

I added the button names, renamed copy to dma, and LVBL_old to LVBLl. There is a convention across the cores that an l suffix stands for last, as it is commonly needed. I made a commit (2e79248a7f3bc28eef6af4f1015fb936d1780451) with these changes.

No big deal, but using the second port of u_vram for DMA would require less multiplexing as data1 and we1 are not used by the GFX.

DMA happens for both tile map and object data then?

gyurco, 2024-03-01:
On the original schematics, it looks like all RAM accesses generates WAIT to the CPU. Implementing this exactly as the original is almost impossible as the SETA chip's real timings are unknown.
The second port cannot be used for DMA, as tilemap copying is usually triggered during active display. Putting tilemap DMA to VBL doesn't work unfortunately. I didn't notice blinking, I'll recheck. When the DMA was slow(er), sometimes there were one-two wrong tiles here and there, but sprites are usually OK.

gyurco, 2024-03-01:
+adding a CEN would require the whole block to be protected by it, otherwise the first step might be missed. I've been there, experimented with it, and the best result was without any throttling.

gyurco, 2024-03-01:
> DMA happens for both tile map and object data then?

Yes, look at in MAME:
https://github.com/mamedev/mame/blob/af6cef00fad3336b27114ba075751558add5e0ab/src/devices/video/x1_001.cpp#L469

It's done together with sprite DMA, but it doesn't work in the core this way, the game doesn't expect it to happen. Also there's a mysterious comment about a cfg bit, I believe writing this should trigger tilemap DMA:
https://github.com/mamedev/mame/blob/af6cef00fad3336b27114ba075751558add5e0ab/src/devices/video/x1_001.cpp#L234

gyurco, 2024-03-01:
Re-checked, it's true there are a few flickers in the bosses with my original code, however with pxl_cen, there are even bad tiles on the background. Just watch attract mode.

jotego, 2024-03-01:
> Re-checked, it's true there are a few flickers in the bosses with my original code, however with pxl_cen, there are even bad tiles on the background. Just watch attract mode.

Granted that it gets worse when making the DMA slow, but the real one will be slow, so we have to keep it slow in the core too. Having it at 6MHz is probably generous. I am always surprised at how often they use a clock divided by 2 or 4 to do these things, rather than using the fastest one available.

I do not see a bus request or wait mechanism coming from the graphics chip. So the CPU must be able to poll its state. This is usually easy to spot if you check access to the GFX chip by the CPU. I am trying to run a simulation to see it, but Kabuki Z seems to hang up in simulation (it can boot in test mode, though).

gyurco, 2024-03-01:
As I see, CPU VRAM access goes through the first custom chip (VRAMWR, MRD), what happens inside it is a mystery. I think until it's figured out, it is better to keep the core working. As it doesn't affect CPU's speed, I speculate it is implementing a time-shared access (so DMA speed doesn't matter until it doesn't conflict with CPU access).

Slowing down the CPU is done by inserting wait states to shared RAM access (SWAIT), DMA has nothing to do with that.

gyurco, 2024-03-01:
I can fabricate an imaginary VRAM access pattern, like:

- CPU slot - Graphics slot - CPU slot - Graphics slot, so CPU has always access, and doesn't conflict with graphics
- DMA is done in blanks only (VBL or HBL), so it also doesn't conflict with graphics (neither the CPU)

Maybe it's possible to measure it on the VRAM bus (as Nemesis on Spritesmind did it with the Genesis chipset), however I'm not the enthusiast who wants to do it :D
But my feeling is that it doesn't affect the CPU speed, so it's not really important.

jotego, 2024-03-01:
If it does not lock the CPU and the software is designed for concurrent access, that makes a lot of sense. And the exact DMA speed would most likely not be relevant.
I have, however, seen cases where the CPU would overwrite parts of the buffer to be copied if let to its own devices. Hopefully, that is not the case here and the CPU will not overwrite data before it gets copied.

gyurco, 2024-03-01:
>  I have, however, seen cases where the CPU would overwrite parts of the buffer to be copied if let to its own devices. Hopefully, that is not the case here and the CPU will not overwrite data before it gets copied.

That's why the tilemap copy is initiated by the CPU (at least its my interpretation), doing it in any fixed point will make the background messy.
And as the current implementation doesn't allow concurrent CPU and DMA access, leaving the DMA running longer will steal access from the CPU (so there are the broken tiles).

gyurco, 2024-03-03:
I tried some strategies, but nothing fixed the barely noticeable boss flickering, so I suggest to just remove the throttling and be happy with it, until somebody measures the real timings of the chip.

jotego, 2024-03-03:
I agree. I think this needs further investigation on the real board. It's awesome that the newer TNZS version works now, though. This has been a good push to the _kiwi_ core. Thanks!

closed 2024-03-03 by jotego