jotego/jtcores #196: CPS1 - Support for Magical Pumpkin

ajgowans, 2023-02-22:
Magical Pumpkin is a recently dumped short theme park game.  it actually runs on the core with the attached .mra file and the mame rom files (hkittymp).  I was able to play through, however the game is meant to use a steering wheel and it is not possible to move all the way to the left or right of the screen with a controller.  

Is it possible to fix so the controller can move the character all the way left and right (potentially add steering wheel support) and release the .mra as part of the core?

[Magical Pumpkin - Puroland de Daibouken (Japan 960712).mra.zip](https://github.com/jotego/jtcores/files/10808543/Magical.Pumpkin.-.Puroland.de.Daibouken.Japan.960712.mra.zip)

jotego, 2023-02-23:
Thank you for the MRA file. I will look into it. It may be related to #150

wwark, 2023-02-25:
Normally Magical Pumpkin used a wheel

jotego, 2023-02-25:
- [X] Added MRA for Magical Pumpkin in https://github.com/jotego/jtcores/commit/64ab7603e110dd3a92c088c4619c7a0742dbfde7
- [ ] Add spinner support

closed 2023-03-01 by jotego

jotego, 2023-03-01:
The game with spinner support is available on JTBIN now.

commit ac61c003 (2023-03-01): Fix #196