jotego/jtcores #200: Chelnov Public Release

jotego, 2023-02-24:


davidromani, 2023-06-10:
First of all many thanks for your precious time & effort to make these cores so incredible. 

Probably this is **not** the right place to ask my doubt, my apologies, but I'm newbie with Mister FPGA and there are so much spreaded information that it's really hard to manage and understand at the begining.

When I try to play this core it seem's that there is a missing file error inside zip. Is this core only available for premium users? Or it's a problem with my instalation?

Thank you so much again and hope that in a future I can help or contribute to this amazing comunity.

[image]

jotego, 2023-06-11:
This core was made public several months ago after the beta phase supported by Patreon members.

The error you are getting is about the files in your ROM dump. You seem to be missing one file there. The core will most likely not work because of that missing file. Note that MAME may still work using that same zip archive because it usually does not need all the files as part of the game logic may be hard coded inside MAME.

closed 2023-06-11 by jotego

davidromani, 2023-06-11:
Thanks for your answer @jotego, you've clarify me some doubts. I'll keep investigating where this file must be located and why is missing.