jotego/jtcores #201: Karnov video measurements

jotego, 2023-02-24:


etriana85, 2023-02-27:
The monitor or video adapter sends a "No Signal" message.
The pcb doesn't work apparently.

etriana85, 2023-03-08:
This pcb won't boot.
New Jotego instructions:
-Measure the CLK and CSYNC, (measurements are attached in images).
-Unsolder chip HMC20_60232_4Z01, location A16.
-Solder socket.
-Mount the CLK on protoboard according to Karnov sch design.
-Measure the output of the osc mounted on the protoboard. (I attach an image of said measurement, with the name of "CLK_test_Karnov"

etriana85, 2023-03-08:
[image]
[image]
[CSYNC_Karnov.webm](https://user-images.githubusercontent.com/92724765/223755704-8b8eb44a-6d34-4f53-ae38-dc31f783c307.webm)

etriana85, 2023-03-08:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

etriana85, 2023-03-08:
Observation:
According to the shc, the CSYNC signal (pin 40 TC176042AF, location 12M) attacks pin 12 of an and gate (location 12E). When following it since the oscilloscope signal was not stable, I detected that said signal does not go out through pin 40, it goes out through pin 41.
Attached images of the sch.

[image]
[image]

etriana85, 2023-05-05:
By Jotego's instruction, the pcb will be prepared to be returned to Atrac17.

closed 2023-05-05 by etriana85