jotego/jtcores #211: jtkiwi: sound issues

jotego, 2023-03-03:
Sound codes 05 and 07 are reported to feel different. An FM instrument is reported to sound lower.

Reported by _Masked Ninja_ on Discord

rogalian, 2023-05-14:
All audio in tnzs is a full octave lower than original hardware.

rogalian, 2023-05-15:
> Could you provide a recording from PCB?

I can't get a direct recording, but I can take a video with a phone?

(MAME audio is correct)

Toryalai1, 2023-05-15:
It is better if it is a direct recording

rogalian, 2023-05-15:
https://github.com/jotego/jtcores/assets/5140184/19cb38c0-1c4f-4488-8db3-cc156ed12887

Best I can manage, should be clear.

jotego, 2023-05-17:
The sound octave problem is fixed now. I think some sound effects may still feel different. If you can report a specific sound code in the test menu that fails for any of the games supported in the core, that will help.

rogalian, 2023-05-17:
It's spot on now, thank you.

closed 2023-06-21 by florgana