jotego/jtcores #213: biocom: Logo screen is not ok for the clone topsecrt2

wwark, 2023-03-03:
With the update of today, the outline of Top Secret is not good:
[image]
Previously we have correct behavior:
[image]

Thanks !

jotego, 2023-03-06:
- the offending layer is SCR2
- it looks fine on level 1

jotego, 2023-07-08:
This behavior comes from the brightness circuit. As far as I can tell, this is correct. As brightness is turned on, even black pixels look gray. This is similar to what happens in some CPS games.

Check how the first row, for a pitch black output, gets values different from 0 as the brightness setting is turned on.

```
Table of colours (voltages)
                  Brightness
     0      1    2     3     4     5     6     7
Colour
0   0.000 0.194 0.412 0.606 0.881 1.075 1.293 1.487
1   0.088 0.282 0.500 0.694 0.969 1.163 1.381 1.575
2   0.194 0.388 0.606 0.800 1.075 1.268 1.487 1.681
3   0.282 0.476 0.694 0.888 1.163 1.357 1.575 1.769
4   0.412 0.606 0.825 1.018 1.293 1.487 1.706 1.899
5   0.500 0.694 0.913 1.107 1.381 1.575 1.794 1.987
6   0.606 0.800 1.018 1.212 1.487 1.681 1.899 2.093
7   0.694 0.888 1.107 1.300 1.575 1.769 1.987 2.181
8   0.881 1.075 1.293 1.487 1.762 1.956 2.174 2.368
9   0.969 1.163 1.381 1.575 1.850 2.044 2.262 2.456
10  1.075 1.268 1.487 1.681 1.956 2.149 2.368 2.562
11  1.163 1.357 1.575 1.769 2.044 2.237 2.456 2.650
12  1.293 1.487 1.706 1.899 2.174 2.368 2.586 2.780
13  1.381 1.575 1.794 1.987 2.262 2.456 2.674 2.868
14  1.487 1.681 1.899 2.093 2.368 2.562 2.780 2.974
15  1.575 1.769 1.987 2.181 2.456 2.650 2.868 3.062
```

The brightness implementation looks correct. I have reviewed the values and I think this is the way it should look like. Probably for the original designers, in their screen, it looked quite dark and didn't notice.

closed 2023-07-08 by jotego

wwark, 2023-07-09:
Thank you for the checking.
On mame it is quite dark on LCD, maybe it is worth to inform them their behaviour may be not correct.

jotego, 2023-07-09:
Yes. I think they implement brightness by multiplying, so if the pixel is zero, the output is zero. But CAPCOM did it by adding so a black pixel becomes gray. That led to this situation in some games.

I don't know if there are scenes that would look bad if I changed the method to multiplying. Maybe it would be worth having an option. 

CAPCOM went this route because it is cheaper as it can be done with two logic chips whereas the other would require three variable gain amplifiers, one per color.

wwark, 2023-07-09:
It is better to keep the correct behavior !