jotego/jtcores #220: flane: missing PCM sounds

jotego, 2023-03-17:
Reported by Alexander StieglMayr on Patreon:

> the new version flane.rbf seems to have a sound issue i.e. when you pick up the girls and the car running through the maze there is no sound at all - other sound effects are ok. I tested it with the former core and mra the sounds are ok (although other issues occur) so it cannot be the hardware setup itself. I also made a comletely fresh installation but there is also the the same issue - no sound when you pickup girls and car is running through maze

Might be related to the MRA generation

jotego, 2023-05-01:
All the sounds that fail have the `pcm_msb` bit set.

jotego, 2023-05-01:
JTFRAME_PROM_START was incorrectly defined and part of the PCM data was not getting loaded into the SDRAM. Fixed in 293965e5

closed 2023-05-01 by jotego

stiegis, 2023-05-01:
Hi Jotego!!!!! Nice work man you are great thanks for fixing Fast Lane and making awesome Arkanoid Revenge of Doh core

your work is highly appreciated  hats off

greetings from Austria

ALEX

commit 293965e5 (2023-05-01): flane: missing PCM sounds fixed. Fix #220