jotego/jtcores #221: Bubble Bobble - falls out of sync with PCB

LordBarker80, 2023-03-19:
I ran the PCB and core next to each other on two almost identical cabs several times. From cold boot around 3.30 minutes in on the demo run through, level 22,  the blue level with the 4 pink monsters, the core desyncs from the original hardware. I'm running the same PCB set as the default Japanese rom. I have repeated the same steps with the same result.

jotego, 2023-03-20:
That's very interesting. Let me review the core and see if there is something suspicious. I designed it a long time ago and cannot reply off the top of my head.

LordBarker80, 2023-03-20:
> That's very interesting. Let me review the core and see if there is something suspicious. I designed it a long time ago and cannot reply off the top of my head.

Thanks Jotego. Happy to send a video with the issue I spotted.

jotego, 2023-03-20:
Please, when you have a minute, send it. Is the core going a bit faster or slower?

jotego, 2023-03-20:
Also noticed that not so long ago I updated the core because there was one thing slowing it down. I assume you're running the newest version in JTBIN.

LordBarker80, 2023-03-20:
> will do. Where can I send it? I'll check speed too.

LordBarker80, 2023-03-20:
> Also noticed that not so long ago I updated the core because there was one thing slowing it down. I assume you're running the newest version in JTBIN.

Yes newest build

jotego, 2023-03-20:
You can either upload here if it is not too long, or to youtube or google drive and share the link

LordBarker80, 2023-03-21:
> You can either upload here if it is not too long, or to youtube or google drive and share the link

Hi Jotego. I have made another video to show the demo sequence. 

I've recorded and synced the sequence three separate times and the behaviour is the same. It 'de-syncs' as the MiSTer BB core plays two levels (at least) differently in the demo sequence to the real board. 

I'm unsure if this is down to the MCU behaving incorrectly/differently or something else entirely.

The speed of the game seems correct on further tests I've done. I need to re-check level 17, for a slowdown comparison. The monster patterns in-game seem correct to me, I've only done a series of quick plays up to around level 25/26 on the MiSTer core, but seems correct. 

This demo sequence change seems odd. The differences start at around 3:15/3:20 and become obvious around 4:45 where the comparisons become different. I'd advise watching the whole video though as you can see the first demo levels behave the same. Thanks, look forward to hearing your thoughts.

Link: https://youtu.be/Lmvx3Bco3dc

jotego, 2023-06-30:
I think that the synchronization problem in the video is most likely related to the crystal oscillator in the PCB ageing. If MiST had gone more slowly, it could have also been related to a CPU slowdown.

Just in case I have made some improvements in the core to reduce the possibility of unintended slowdowns. I will update the binaries later today.

closed 2023-06-30 by jotego

jotego, 2023-10-22:
Another accuracy improvement done in ce66884826a148476f79ad97e753d0fd06086203