jotego/jtcores #223: The New Zealand Story music

kold669, 2023-03-24:
Don't know if it's a known bug but The New Zealand Story music sounds different then the original pcb or mame.

I could find only these videos as reference.

Mister FPGA:
https://www.youtube.com/watch?v=VSFivSVQv2I

Mame or arcade pcb:
https://www.youtube.com/watch?v=91VJQKWc29w&t=53s

limi, 2023-05-16:
> I love the fact that we now have The NewZealand Story on Mister but I have noticed that music and sound effects are very low pitch. A good example is when you collect the bouncing fireballs it does not sound right.

From this thread:
https://misterfpga.org/viewtopic.php?t=6520

jotego, 2023-05-16:
Thank you for the information. This is already reported in #211 so I will close this one.