jotego/jtcores #226: Super Hangon: pixel jitter

JanGunheD, 2023-03-24:
Looking closely, all pixels jitter a tiny bit around. This becomes especially apparent when combined with filters like shadow masks.

A similar thing happened with the first split versions of the Xain'd Sleena core (just in case that might help).

Using vsync_adjust=2 (just in case this only happens in this low latency mode).

jotego, 2023-03-25:
This is due to the MiSTer HDMI subsystem using a different sampling frequency from the core pixel frequency. It happens on many cores as the pixel clock in arcade games is rather arbitrary -there was no standard for CRTs- and the HDMI subsystem has not been designed to cope with this.

This is a problem that involves fixing all the video signal chain in MiSTer. I cannot tackle it because it would stop core development for several weeks. The designers of that part of the MiSTer system are the ideal people to work on this.

A similar problem happens in the audio signal chain, although it's a bit less noticeable because people are nowadays used to the MAME sound, which is also poorly done in this regard.

closed 2023-03-25 by jotego

JanGunheD, 2023-04-26:
It seemed to be solved for a while, now with the latest updates it is happening again.