jotego/jtcores #227: outrun: large objects may show a partially blank line on the left side

jotego, 2023-03-25:
Seen in 5b3830c6
- Related to the countdown option for objects?
- Related to the line-based frame buffer?

jotego, 2023-03-25:
The blank lines appear at the same relative horizontal position in the object. Varying the horizontal zoom shows how the blanks follow the object and can appear closer to the border or the screen centre depending on the h-zoom setting.
The point at which the blanking turns to solid object is not fixed... bad data from the rom?

jotego, 2023-03-25:
- not related to the safe re-start in: https://github.com/jotego/jtcores/blob/a4b5b39a2a3238168ef84ad4a49cfd28db0f2e87/cores/outrun/hdl/jtoutrun_obj_draw.v#L148

jotego, 2023-03-27:
The problem occurs in the frame buffer. Some lines are not drawn because the request occurs too lat. Making `hlim` larger has a clear effect. `hblen`is $4F, by making:

`hlim   <= hcnt - (hblen-st_addr);`

It is possible to stabilize the lines gaps vertically setting `st_addr=$8`. The problem cannot be completely removed with this approach, though.

ksdomino, 2023-09-24:
This might not be a fixable bug. Outrun arcade can only " display 76 scenery sprites at any one time" so some arches are broken in the arcade version. source: https://reassembler.blogspot.com/2012/01/gateways-broken-arches.html

jotego, 2024-01-14:
Limiting it to 76 does not solve the issue as that limitation came from the software side of the system, not the hardware (the FPGA in our case). I think I'll settle with the current approach until an FPGA that enables a proper frame buffer shows up.

closed 2024-01-14 by jotego