jotego/jtcores #233: pocket: 19xx, Dimahoo, flip settings

jotego, 2023-03-27:
Apparently the image is upside down with the default settings

YLFAndy, 2023-04-23:
Setting option 3 in cores/jotego.jtcps2/video.json to 270 fixes all of the flipped games and doesn't appear to impact any of the non-flipped ones. I haven't tested fully, but it solved the issue for me in 19XX and Dimahoo both and every other game I've tried has not had any negative impact.

jotego, 2023-09-08:
Although it can fixed through the JSON file, this is actually related to missing the default values in the NVRAM as the beta version does not load them on the Pocket.

jotego, 2023-09-08:
This seems to already be working on commit 1e5a4916. I am not sure which commit fixed it.

jtmiki, 2023-11-02:
One patreon wrote this today that Dimahoo's screen orientation has changed abnormally again.

jotego, 2024-01-14:
Everything seems fine in f8166fbb. It was a false alarm.

closed 2024-01-14 by jotego