jotego/jtcores #234: Splatter House schematics

jotego, 2023-04-03:
Extract schematics for the Splatter House PCB in the **cores/mania/sch** folder.

The list of games for this system is [here](http://system16.com/hardware.php?id=524). Check for which other games we have a PCB and compare them.

etriana85, 2023-04-03:
I'll work on this

etriana85, 2023-05-05:
Checking the list of known games for the Namco 1 system, I saw that Pac-Man was on the list. So I looked up the Pac - Mania schematic and compared it to the Splatter House board. I identified that both have the same programming.
Attached schematic link of Pac - Mania, for more information.
https://www.arcade-museum.com/manuals-videogames/P/PacManiaSP3151stPrintingSchematicPackage.pdf

closed 2023-05-05 by etriana85