jotego/jtcores #235: Golden Axe US Set 6 (default MRA) starts in Service Mode, no Test button mappable to exit.

YLFAndy, 2023-04-04:
As discussed on the Discord, this issue has previously been filed but was closed. The default Golden Axe MRA - Golden Axe (set 6, US) (8751 317-123A).mra - starts in Service Mode with no way to exit as the Test button is not wired up. 

Repro Steps:
Fresh MiSTer install
Update_all.sh with Arcade cores enabled
Launch the OSD and navigate to Arcade > Golden Axe (set 6, US) (8751 317-123A) in the main arcade list (alternates work)

Result: The core launches directly into Service Mode as if the DIP is thrown in the initial memory write.

Expected: The game to launch into standard mode with default settings if no other settings exist.

This can sometimes be worked around by toggling any DIP, then returning to the main core menu and toggling Service Mode quickly. 

https://youtu.be/zecEnmqrYUQ

YLFAndy, 2023-04-04:
This is linked to https://github.com/jotego/jtcores/issues/171

Which was closed when I last looked at it but has been reopened, with no details apart from 'related to MCU core consolidation.'

vanfanel, 2023-05-05:
I can confirm this issue is still happening (after a year or more) with latest core and US Set 6 roms.

florgana, 2023-06-21:
duplicate in #171

closed 2023-06-21 by florgana