jotego/jtcores #240: Outrun core dip switch option not working.

suww37, 2023-04-15:
Outrun core dip switch option not working.

jotego, 2023-05-02:
For some reason, the OSD screen showing DIP switches and other options is not displayed on MiSTer. Only the core switch screen gets shown. Seen in f925873f but it must have started much earlier than that.

asturur, 2023-09-14:
Just an heads up i tried the lastest compiled core i found in the github action and this issue seems solved.
(the red car logo splash screen of outrun is now flashing red and blue pixels, but that could be my mister)

jotego, 2023-09-15:
I guess you tried the version compiled by the GitHub action... thank you.

The OSD menu operation is restored in f43b2b5f. This bug was very weird. It might have been related to the relaxed timing constraints for the MiSTer framework used on some compilations.

closed 2023-09-15 by jotego