jotego/jtcores #243: outrun: PCM instruments may sound when they shouldn't

jotego, 2023-04-19:
Moving the PCM scratch address to $87 instead of $83 fixed the awful car noises (#166) but some percussion instruments still sound when they shouldn't. Are samples running passed the end?

jotego, 2023-05-01:
The Z80 is in the clk24 domain, but it has RECOVERY set. Disabling it seems to fix the problem:

```
-jtframe_sysz80 #(.RAM_AW(11)) u_cpu(
+jtframe_sysz80 #(.RAM_AW(11),.RECOVERY(0)) u_cpu(
```

jotego, 2023-05-01:
Fixed in 86b959bd

jotego, 2023-09-26:
After releasing it, used feedback is that it sounds better, but there are still instances of wrong sounds

jtmiki, 2024-02-09:
First, I verified music 1, music 2 and music 3 in the sound test by listening to the percussion-only performance. 
And then I listened to the performance with the melody and verified it further, but there was nothing like noise and there seemed to be no problem with the way the sound was heard. 
I also selected each of the three musics and verified them while playing, but did not find any sound problems. 
The version I tested was Out Run (deluxe sitdown) (Enhanced Edition vl.0.3)

music 1. Magical sound shower
music 2. Passing Breeze
music 3. Splash Wave

TulioAdriano, 2025-02-14:
Hi, after speaking with @jotego, April last year, I reported that the sound issues in Out Run PCM still remain. I finally was able to get my hands on an Out Run board, and I made the recording of Splash Wave to exemplify the issue. 

The core used, was the one normally listed in the Arcade section (Sitdown-Upright, Rev B), recorded today (Feb 14, 2025) with the last update on MiSTer being done 3 days ago. 

Please see the recordings:
[Splash Wave (Mister)](https://www.dropbox.com/scl/fi/qwv45sh18ge82en7srs2h/Splash-Wave-MiSTer.wav?rlkey=9vvrgafm87g7um03new3styqd&st=4kwfvmie&dl=1)
[Splash Wave (Arcade PCB)](https://www.dropbox.com/scl/fi/1vo4equvue9dmre13t1ln/Splash-Wave-audio-PCB-mod.wav?rlkey=a1xee0o2tytbhn8pt1sfts0lk&st=3zv1iplq&dl=1)

Note that I made a small audio mod on this PCB to bypass the terrible filter chips and enhance the quality of the analog path. An original recording pre-mod was made [here](https://www.dropbox.com/scl/fi/ckewlcdjpwmqoxgo5gl75/Splash-Wave.wav?rlkey=zfk6bnsipq2rr3vuafedrib4o&st=7bjq1t95&dl=1) (protect your ears).

The issues are (not in specific order):
1. Missed Beats
2. Beats triggered where they shouldn't
3. Beats not being cut out 

All of these issues can be heard within the first 20 seconds of the recording, but they do occur all throughout it. 

If there are any questions, please let me know.

jotego, 2025-02-15:
@jtmiki try it again with today's (Monday 17th Feb) core.

jtmiki, 2025-02-17:
I have tested it with today's core. (Commit: df8c5e7)
The sound quality of Splash Wave seems closer to the original one.
However, it would be better if the melodies were louder.
The intro part of it, I don't know what instrument it is, but the hand clapping sound is missing.

commit 10beffbf (2023-05-01): outrun: PCM fixed #243